MicrosoftDocs/windows-driver-docs #629: Request to change the details for altitude values 323000 and 327000

cyren-oschneider, 2018-07-04:
Hi,

I am currently the maintainer of the driver in question for altitude 327000 and used to be the maintainer of the one at 323000 before that product was abandoned. Authentium sold the AV business to Commtouch years ago and we're now called Cyren. Also, the driver is named `amp.sys` instead of the `avmf.sys` mentioned in the documentation.

We've kept using that exact same altitude value ever since, despite the ownership change and the apparent name change of the driver (personally I wasn't aware of the old name).

Also, since I am at it, please correct the entry for altitude 323000. Commtouch (at the time), now named Cyren, bought FRISK Software at the end of 2012 and thus now owns F-PROT and the `FPAV_RTP.sys` mini-filter as well. The entry "f-protect" was, by the way, always wrong and should have been F-PROT or FPAV (our acronym for F-PROT Antivirus).

```
fpav_rtp.sys | 323000 | Cyren (F-PROT)
amp.sys | 327000 | Cyren (CSAM)
```
The part in parentheses would be the product name.

You can reach me via `$firstname@cyren.com` or `$firstname.$lastname@cyren.com` or get in touch with us via the Microsoft Virus Initiative (MVI) of which we're a member to confirm and verify the details.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 0a2d558e-1073-85d5-f863-5bdbf6e3ddc6
* Version Independent ID: 62e4d735-803f-7608-154b-7032dc1ceb3d
* Content: [Allocated Altitudes](https://docs.microsoft.com/en-us/windows-hardware/drivers/ifs/allocated-altitudes)
* Content Source: [windows-driver-docs-pr/ifs/allocated-altitudes.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/ifs/allocated-altitudes.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-07-05:
Thanks, @cyren-oschneider we can change it in the docs, and we'll get you in touch with the right person to propagate any approved changes.
@aahill please shepherd this issue to the right dev or PM.

EliotSeattle, 2018-07-05:
Related to Issue 630

assarbad, 2018-07-11:
As per the suggestion from @aahill in issue #630 I'll try to use the form as given and see how far it gets me. Thanks so far.

EliotSeattle, 2018-10-10:
@assarbad is this still an issue?

assarbad, 2018-10-15:
@EliotSeattle nope. Same person, btw, just ended up being logged on with the wrong credentials accidentally.

EliotSeattle, 2018-10-15:
Happens to the best of us. Glad this is resolved. Thanks so much!

closed 2018-10-15 by EliotSeattle