MicrosoftDocs/windows-driver-docs #630: Please document a means to update details for an existing altitude value

cyren-oschneider, 2018-07-04:
In issue #629 I was requesting the details to two altitude values to be updated.

The reason I posted it here as feedback was, that [the article](https://docs.microsoft.com/en-us/windows-hardware/drivers/ifs/minifilter-altitude-request) against which I am filing this ticket doesn't mention **how to request such data to be updated**. Besides, it stands to reason that neither of the email addresses that were originally given in the altitude assignment request exist any longer. Or, technically, for the address on the Authentium company domain that may well be the case, but Cyren took ownership of the AV business division from Authentium, not of the domain. So we would not be able to access email sent to this old email address, even if it still happens to exist.

As for the old `frisk.is` and `f-prot.com` email addresses I know that many of them became defunct over the years and that affected in particular role aliases like those established over a decade ago as contact address when initially requesting the altitude assignment. And last but not least I would literally have to guess what email addresses were used in the original communication. That seems impractical.

The one thing I _can_ tell you, though, is that none of the original developers of those mini-filters are with the company anymore. For the code originating with Authentium I even only know the name of the guy, for the code from the FRISK side we worked together until about five and a half years ago.

So, it would be great if you could document a method for a company to re-assert ownership over an already assigned altitude value in order to update the details for that assignment.

Thanks and best regards,

Oliver

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 3f62aea8-127c-317f-07d8-6f6d93df7bb1
* Version Independent ID: e85bfe16-3267-7aee-65c8-46c59bae4a6f
* Content: [Minifilter Altitude Request](https://docs.microsoft.com/en-us/windows-hardware/drivers/ifs/minifilter-altitude-request)
* Content Source: [windows-driver-docs-pr/ifs/minifilter-altitude-request.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/ifs/minifilter-altitude-request.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-07-05:
Thanks, @cyren-oschneider we can change it in the docs, and we'll get you in touch with the right person to propagate any approved changes, just like Issue 629.
@aahill please shepherd this issue to the right dev or PM.

aahill, 2018-07-11:
Hi @cyren-oschneider, thanks for your patience. I checked with the development team, and they said that you can use the same form on the [page](https://docs.microsoft.com/en-us/windows-hardware/drivers/ifs/minifilter-altitude-request) you mention. Just be sure to include these details in the "Additional information" section. From there, the team should be able to help you out.

I agree that this topic could be much clearer - I'll update it and have a revised version with this information (and some other quality of life updates) shortly.

Thanks!

assarbad, 2018-07-11:
@aahill will try to send an email that way then. Need to figure out what contact address best to use, though. Thanks!

aahill, 2018-07-19:
Hello @assarbad, and @cyren-oschneider, were you able to update your altitudes via email?

cyren-oschneider, 2018-08-06:
Still waiting for our IT, but I _will_ contact your team as soon as things are sorted out.

lorihollasch, 2019-04-25:
@ghost, I am going to close this issue. Please email fsfcomm@microsoft.com if you still need to update your altitude information!

closed 2019-04-25 by lorihollasch