MicrosoftDocs/windows-driver-docs #631: How to remove the node "General Software Component" that create from extension INF

lixiaogangkkk, 2018-07-05:
Dear MS Guys，
    I'm working on having an UMDF driver to upgrade firmware for my device. 
    1. Now I create an extension INF to call AddComponent directive, and found there is a software component node created after running the extension INF. (the node is associated with my real device node)
    2. I can install UMDF driver to this software component node to do firmware upgrading operations.
    3. I uninstall drivers of software component node and driver of my real device node from device manager (also select to delete driver from drive repository).
    4. After stage 3, I detach and re-attach my device to host, there software component is still in device manager.
    Question: How could I remove this node from device manager?

Thanks,
Synaptics Inc, Steven Li

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 61e5e417-647c-0c3a-9c25-83a7be9efe3a
* Version Independent ID: 6ccb6634-97cb-b66f-3ca5-0e58218abdc0
* Content: [Using an Extension INF File](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/using-an-extension-inf-file)
* Content Source: [windows-driver-docs-pr/install/using-an-extension-inf-file.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/using-an-extension-inf-file.md)
* Product: **windows-hardware**
* GitHub Login: @tedhudek
* Microsoft Alias: **windowsdriverdev**

NabilFates, 2018-07-16:
Hey Steven,

Is it possible to provide some more details on why you want to remove the component device node? 

Thanks,
Nabil

tedhudek, 2018-07-18:
@lixiaogangkkk please find Nabil's reply above

tedhudek, 2018-07-19:
Closing due to lack of activity, feel free to reopen if needed

closed 2018-07-19 by tedhudek

lixiaogangkkk, 2018-07-25:
Hi Nabil,
     Our QA team just run normal installation/uninstallation as they do for device driver. That's why I post my question here. But I have got answer from MSFT partner dashboard. 

Thanks,
Steven

lixiaogangkkk, 2018-08-21:
Dear MS Guys，
I got a new question about the extension INF that combined with my firmware upgrade UMDF driver.
   1. We have a customized device driver for my USB audio device
   2. I create an extension INF to extend a component node, I will install my firmware upgrade UMDF driver onto this node and driver will do firmware upgrade after device Plugin.
   Question: 
   1. Should I have to pack the extension INF together with my device driver? 
   2. Can I seperate this extension INF as a single WHQL submission then do WU?
   3. Should I keep extension INF the same version as device driver INF?

Thanks,
Steven

NabilFates, 2018-08-21:
Hey Steven,

You don't need to package the extension INF with the firmware upgrade INF. You can also publish them separately to WU. As a matter of fact, if this firmware upgrader driver behaves the same on all the pieces of hardware that it applies to, it is preferable to keep them separate, so that they're properly componentized. The extension INF and device driver can have separate versions, and you may set them at your discretion.

Thanks,
Nabil

lixiaogangkkk, 2018-08-24:
Thanks, Nabil. All become clear to me.