MicrosoftDocs/windows-driver-docs #632: What do the defines in part 3 mean?

karel-3d, 2018-07-06:
What do all the defines in part 3 mean? What is `#define SIGNERSCORE_LOGO_PREMIUM`? What? Where do I add this - to my inf file, to my driver file?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 70062259-782d-7d6a-08a1-d01d33edf46c
* Version Independent ID: 2922b3e2-ae65-2a09-8de3-fa80808ff953
* Content: [Driver matching criteria](https://docs.microsoft.com/en-us/windows-hardware/drivers/display/driver-matching-criteria)
* Content Source: [windows-driver-docs-pr/display/driver-matching-criteria.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/display/driver-matching-criteria.md)
* Product: **windows-hardware**
* GitHub Login: @nabazan-microsoft
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-07-09:
Thanks for the questions, @karel-3d 
@mimisasouvanh can you please triage this request and add it to your backlog appropriately?

karel-3d, 2018-07-10:
@EliotSeattle What would I love to see are more clarifications in the text itself :) right now it is very confusing.

EliotSeattle, 2018-07-10:
@karel-3d Do you mean the section you spoke of, this particular topic, our docs in general? Also, for this particular topic, I must point out that WDDM 1.2 was released in Window 8 and is not our most recent offering. Are you also targeting Windows 8?

karel-3d, 2018-07-11:
Hello,

I am developing a driver that has to work in win 7, win 8 and win 10.

The problem is that on some PCs, some other drivers are "overriding" our drivers on our devices. So I tried to find an order how is Windows deciding which driver to use for a given device.

The only document that describes this is the linked one. However it is not very clear what the linked document actually says and what do the defines actually mean, and if it even solves my problem in the first place.

karel-3d, 2018-07-11:
...oh now I noticed it is in display drivers, so not at all related. My device is a USB device not related to displays. Should I close this issue?

karel-3d, 2018-07-11:
What I needed was this document

https://docs.microsoft.com/en-us/windows-hardware/drivers/install/overview-of-the-driver-selection-process

closed 2018-07-11 by karel-3d