MicrosoftDocs/windows-driver-docs #633: Old SDV Blog Link

druske, 2018-07-09:
The latest SDV blog content is over 8 years old; I guess parts 3 through 6 of the 6 article series are running a little behind schedule. It'd be nice to have some fresher content, or at least some release notes on more recent SDV changes.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 1173541b-25be-4821-69ac-3784107fb215
* Version Independent ID: 53df4de7-78ca-9f67-1528-6fc116497524
* Content: [Static Driver Verifier](https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/static-driver-verifier)
* Content Source: [windows-driver-docs-pr/devtest/static-driver-verifier.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/devtest/static-driver-verifier.md)
* Product: **windows-hardware**
* GitHub Login: @nabazan-microsoft
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-07-09:
Thanks for pointing this out, @druske 
@aahill can you please check with our friends on the DV team to see what they've replaced that content with?

EliotSeattle, 2019-07-12:
This was fixed quite a while ago, but I missed updating the Issue. I'm so sorry. This page now links to the  Windows Hardware Certification content (https://techcommunity.microsoft.com/t5/Windows-Hardware-Certification/bg-p/WindowsHardwareCertification) at its new location.

Eliot

closed 2019-07-12 by EliotSeattle