MicrosoftDocs/windows-driver-docs #634: Confusing text

kildonan5, 2018-07-09:
Under Universal INF the text states "If you DO NOT get any errors or warnings with error number 130x and error text ("Found legacy Xxx operation..."), your driver INF file is universal."  Should this say your drive INF "is NOT universal" or is the given text accurate?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 48275f87-4086-1ca5-a190-c741f29ad348
* Version Independent ID: 5481885d-3063-527b-a85e-6a898dbcb990
* Content: [INF Validation Errors and Warnings](https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/inf-validation-errors-and-warnings)
* Content Source: [windows-driver-docs-pr/devtest/inf-validation-errors-and-warnings.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/devtest/inf-validation-errors-and-warnings.md)
* Product: **windows-hardware**
* GitHub Login: @nabazan-microsoft
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-07-09:
Thanks for the question, @kildonan5 
@tedhudek can you please make the text less ambiguous?

tedhudek, 2018-07-10:
Thanks @kildonan5, I've updated the page.  Please let us know if you still have issues.

kildonan5, 2018-07-10:
Hi @tedhudek thank you, but I'm still confused because the Microsoft documentation on [Universal INF file](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/using-a-universal-inf-file), explicitly states; 

```
The INF AddReg Directive is valid if entries in the specified add-registry-section have a reg-root value of HKR, or in the following cases:

For registration of Component Object Model (COM) objects, a key may be written under:
HKCR
HKLM\SOFTWARE\Classes
For creation of Hardware Media Foundation Transforms (MFTs), a key may be written under:
HKLM\SOFTWARE\Microsoft\Windows Media Foundation
HKLM\SOFTWARE\WOW6432Node\Microsoft\Windows Media Foundation
HKLM\SOFTWARE\WOW3232Node\Microsoft\Windows Media Foundation

```
I modified an INF file, and specified an AddReg value outside of those allowed in the list above (e.g. I added a custom key under HKLM). Subsequently, the driver install logs included the error code "1305: Found legacy operation using non-relative key". The error code 1305 was found on this document, and yet the document states " Your driver INF file is universal if you do not get any errors or warnings with error number in the range 130xx." So which is it?

tedhudek, 2018-07-10:
Hi @kildonan5, I'm not entirely clear, maybe you can help me understand. The page you quote above about AddReg suggests to me that the custom key you added under HKLM would not satisfy the conditions listed. So your INF would be non-universal because of the custom key. So at compile time you see error 1305, which means non-universal. I don't see the contradiction.

tedhudek, 2018-07-10:
BTW, I had a typo when I put `130xx` instead of `13xx`.  I'll fix that, but it doesn't seem related to your question.

kildonan5, 2018-07-11:
So by virtue of my INF failing to conform to the universal INF file rules, it is by definition, non-universal. In my mind, it would make more sense to say I have a universal inf file with invalid sytnax, but in anycase I understand what this articles message is getting at now. Perhaps if nothing else this issue will serve to clarify the issue for any future 'googler' (binger?) that are confused, as I was.