MicrosoftDocs/windows-driver-docs #639: Broken Links in the document

asachida, 2018-07-11:
DispatchPnP ( https://msdn.microsoft.com/library/windows/hardware/ff543341) and DispatchPower (https://msdn.microsoft.com/library/windows/hardware/ff543354) seems to be broken.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 0d77daf6-7e6d-e045-a300-588c718bd972
* Version Independent ID: 2535107c-d683-71dd-e17f-468d1835f55a
* Content: [Introduction to PSHED Plug-Ins](https://docs.microsoft.com/en-us/windows-hardware/drivers/whea/introduction-to-pshed-plug-ins)
* Content Source: [windows-driver-docs-pr/whea/introduction-to-pshed-plug-ins.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/whea/introduction-to-pshed-plug-ins.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-07-11:
Thanks @asachida for pointing this out.
@PageWriter-MSFT this seems to be more than this page. The original rid was kernel.dispatchpnp, but that no longer exists in SD. The broken links appear in more topics than this. For instance:
- whea/introduction-to-pshed-plug-ins
- kernel/dispatchpnp-routines
- kernel/irp-mj-pnp
Can you please find the correct URL? Then we can look at a find/replace.

PageWriter-MSFT, 2018-07-19:
Hello,

Both DispatchPnP and DispatchPower implement the DRIVER_DISPATCH callback routine. I have updated the link.

Thanks for finding the broken link.

Page Writer

closed 2018-07-19 by PageWriter-MSFT