MicrosoftDocs/windows-driver-docs #641: Mr. Keith Styles

ozhuntsman, 2018-07-12:
Error code 28 is useless. It's no help ! To tell a user the device is unknown without telling them a solution to the problem, is a waste of time. There has to be a better way because most users don't have a clue !

---
#### Document details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: a246b56c-e11e-f80d-92e6-e4d84e593ef7
* Version Independent ID: dfc9b9f7-2a8b-35f3-8328-67d9331ed2e6
* Content: [CM_PROB_FAILED_INSTALL](https://docs.microsoft.com/en-au/windows-hardware/drivers/install/cm-prob-failed-install)
* Content Source: [windows-driver-docs-pr/install/cm-prob-failed-install.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/cm-prob-failed-install.md)
* Product: **windows-hardware**
* GitHub Login: @tedhudek
* Microsoft Alias: **windowsdriverdev**

tedhudek, 2018-07-17:
Thanks @ozhuntsman really the only advice I have for you is to check the company's website for the latest drivers for the device.  I updated the doc page to indicate this.  Best of luck with this issue!

closed 2018-07-17 by tedhudek

ozhuntsman, 2018-07-17:
If I knew what the device was, That's the first thing I'd do !  Windows 10 just tells a user it's an unknown device !!!
To tell a user its an unknown device (because Windows 10 is dumb) and not identify the device, so a user can check the owners web sight is just plain dumb !

tedhudek, 2018-07-17:
Can you provide more details on where you encountered the error and what you were trying to do?

ozhuntsman, 2018-07-18:
Hi Ted,
The problem isn't complicated.
Prior to doing a clean instal of windows there were NO unknown drivers
listed in the device manager hardware list. After reinstalling windows
v1803 (a clean instal), windows has forgotten several drivers.

They are listed in the device manager as "UNKNOWN". It makes no sense to
tell a user the hardware is unknown when it plainly was known prior to the
V1803 installation !

I do understand that windows may not know because the Registry has been
updated, but that's no excuse to leave a user scratching his head and
wondering what is the hardware that windows did know about but now doesn't
!

My Sony VAIO with a touch screen, isn't that old and hasn't had a lot of
use, so I'm not about to toss it into the trash and buy a new one, just
because Microsoft has decided it isn't going to keep track of drivers for
it !

If I knew what the hardware WAS, I'd go to the manufacturers site for
drivers. But Windows doesn't identify the hardware ! Windows cops out, or
rather the programmer is lazy and fails to keep track of the hardware.
Every item of hardware has an ID, so why isn't it reported. With that piece
of info, a user can do a lookup on line to identify the hardware !

I'll reiterate...Windows DID know about it prior to installing V1803 !

regards,
..Keith

tedhudek, 2018-07-18:
Hi @ozhuntsman, 

I am truly sorry for the troubles you’re going through. We create the documentation for companies that undertake driver development and deployment from a commercial perspective, and we’re staffed to solve issues with the supporting documentation. The documentation page that you created the issue from is for commercial use in deploying drivers. For consumer driver issues, your best bet is to look at the Microsoft Help topics that are on support.microsoft.com. After poking around a little, you may be able to directly find a solution or use the help agent on the site. I found the topic,  Update drivers in Windows 10 (https://support.microsoft.com/en-us/help/4028443/windows-update-drivers-in-windows-10), that looks like it might be in line with what you’re looking for.

Sorry that I can’t offer you something more specific than that.