MicrosoftDocs/windows-driver-docs #643: Please extend this by details on the lifetime of an ELAM driver

cyren-oschneider, 2018-07-12:
Could you please add information on when the ELAM driver gets unloaded and how to ensure that the handoff of state information to the user mode component(s) happens _prior_ to ELAM being unloaded?

For example while **Registry callbacks** are mentioned, I fail to see the use, given that the ELAM will be unloaded. So these callbacks don't seem to be useful for the uptime of the machine, but just for a very short time frame.

Thanks and best regards,

Oliver

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: da63a995-c445-b722-73d7-2379d00ad1d5
* Version Independent ID: eef47bcf-ed60-ee4b-9341-fef0c8d1eb1d
* Content: [ELAM Driver Requirements](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/elam-driver-requirements)
* Content Source: [windows-driver-docs-pr/install/elam-driver-requirements.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/elam-driver-requirements.md)
* Product: **windows-hardware**
* GitHub Login: @tedhudek
* Microsoft Alias: **windowsdriverdev**

tedhudek, 2018-07-18:
Hi @cyren-oschneider 
The ELAM driver can handover to another boot start driver that then starts monitoring drivers and user mode.

Working on getting more info for you but that's all I have for now!

tedhudek, 2018-07-19:
Updated the doc page a bit more. Hope it helps, feel free to reopen if not

closed 2018-07-19 by tedhudek

cyren-oschneider, 2018-07-31:
Thanks, that clarifies it a bit, @tedhudek !