MicrosoftDocs/windows-driver-docs #644: Cannot download any of the old WDK versions (8,8.1, 10 1607, 10 1703), broken installer

jaimeff, 2018-07-12:
The installer does download correctly, but once executed it tries to download the msi file and fails. 

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 3f5d6d55-cb35-6b9e-f8ca-e75f70826dc3
* Version Independent ID: 1d913516-b1e0-e872-c337-e30ed5dffce8
* Content: [Previous WDK versions and other downloads](https://docs.microsoft.com/en-us/windows-hardware/drivers/other-wdk-downloads)
* Content Source: [windows-driver-docs-pr/other-wdk-downloads.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/other-wdk-downloads.md)
* Product: **windows-hardware**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-07-12:
Hi @jaimeff 
I can't repro this. All of those downloads work for me. Will try on a couple of other machines before I close this, though.

EliotSeattle, 2018-07-17:
Works just fine. Is it still a problem for you?

closed 2018-07-17 by EliotSeattle