MicrosoftDocs/windows-driver-docs #645: How to use measured boot with ELAM

cyren-oschneider, 2018-07-12:
The article mentions [`Tbsi_Revoke_Attestation()`](https://docs.microsoft.com/en-us/windows/desktop/api/tbs/nf-tbs-tbsi_revoke_attestation) and gives an indication that the ELAM can call it, although the `WINAPI` in the function prototype seems out of place.

It would be nice to link to the documentation for that function.

However, at this point the documentation for `Tbsi_Revoke_Attestation()` doesn't even mention its availability in kernel mode. Alas, the `tbs.h` is in the `shared` folder in WDK/SDK 10.0.17134.0 and a `tbs.lib` exists in `km`.

Since the documentation for that function hasn't been migrated over to GitHub I don't know how to suggest updating its documentation to include the kernel mode scenario.

Another piece of information that I found missing was under what circumstances to call `Tbsi_Revoke_Attestation()` from the ELAM. Just call it unconditionally when a rootkit was detected or am I supposed to detect in some way that measured boot is activated and only call this function whenever it is?

Thanks and best regards,

Oliver

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: da63a995-c445-b722-73d7-2379d00ad1d5
* Version Independent ID: eef47bcf-ed60-ee4b-9341-fef0c8d1eb1d
* Content: [ELAM Driver Requirements](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/elam-driver-requirements)
* Content Source: [windows-driver-docs-pr/install/elam-driver-requirements.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/elam-driver-requirements.md)
* Product: **windows-hardware**
* GitHub Login: @tedhudek
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-07-12:
Thanks, @cyren-oschneider 
@tedhudek can you look into this please?

tedhudek, 2018-07-18:
Hi @cyren-oschneider
You should call Tbsi_Revoke_Attestation()anytime the ELAM agent detects something malicious e.g. a rootkit. You’ll get an error if there is a problem with measured boot, e.g. no TPM on the system. Measured boot is always on afaik if there is a TPM.

Tbsi_Revoke_Attestation  is callable from kernel mode.

Does that help?

tedhudek, 2018-07-18:
Also, I'm working to add the kernel mode callable caveat to the ref page for the function.  Stay tuned for that

closed 2018-07-19 by tedhudek

cyren-oschneider, 2018-07-31:
Hello @tedhudek that helps, thanks. I was out of office, so that's why I respond so late.

One question though: so if/when I encounter an error code from `Tbsi_Revoke_Attestation()` I just ignore it? At the early stage ELAM is running, I hardly have any other alternatives, right?

Thanks,

Oliver