MicrosoftDocs/windows-driver-docs #650: Table is broken

sahikon, 2018-07-13:
The table that supposed to be after the first paragraph is not rendered properly.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: ff3034ee-2707-7672-edf8-a963b09f90e4
* Version Independent ID: b9732fef-bbb4-6d55-1234-e2a9d1512cc7
* Content: [Bluetooth Core Driver Layer and Supported Power Transitions](https://docs.microsoft.com/en-us/windows-hardware/drivers/bluetooth/bluetooth-core-driver-layer-and-supported-power-transitions)
* Content Source: [windows-driver-docs-pr/bluetooth/bluetooth-core-driver-layer-and-supported-power-transitions.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/bluetooth/bluetooth-core-driver-layer-and-supported-power-transitions.md)
* Product: **windows-hardware**
* GitHub Login: @nabazan-microsoft
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-07-13:
Thanks, @sahikon 
I'll see if I can get to this today.

EliotSeattle, 2018-07-13:
Table is created and is in the publishing pipeline to go live. Thanks!

closed 2018-07-13 by EliotSeattle