MicrosoftDocs/windows-driver-docs #652: Documentation incomplete

LuigiBruno, 2018-07-13:
Please add more information about this bug check error, its possible causes and resolution techniques.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 4937e23b-c051-74c2-7299-8e07e716363b
* Version Independent ID: bcce8bfb-3563-cebd-55f2-39e2c139fbf1
* Content: [Bug Check 0x1B PFN_SHARE_COUNT](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x1b--pfn-share-count)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0x1b--pfn-share-count.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0x1b--pfn-share-count.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2018-08-18:
@LuigiBruno - I'm checking with the dev team to see if we can provide more information here.

I did find a blog entry specific to VMWare on this stop code:
http://blog.jmay.us/2016/06/pfnsharecount-bsod-in-vmware-solved.html

DOMARS, 2018-09-15:
@LuigiBruno - I checked with the development team and they confirmed that this stop code has not been used by the Windows operating system itself for a very long time. Because of this I'm not able to provide any additional information on what this failure means or specific the use of the parameters.  In this case the stop code is in use by VMWare as described in this blog. 
http://blog.jmay.us/2016/06/pfnsharecount-bsod-in-vmware-solved.html
They may be able to provide additional information on their use of this stop code. Reusing a Windows stop code like this by an external partner is unusual, but it is possible. Thanks for this request and your interest in improving the stop code documentation.

closed 2018-09-15 by DOMARS