MicrosoftDocs/windows-driver-docs #654: No mention of user mode driver signing process

nirvantosh, 2018-07-13:
Hi, 

The explanation above states:

"Starting with Windows 10, version 1607, Windows will not load any new kernel mode drivers..."

This is clear in terms of kernel mode drivers, but what about user mode drivers, will they also need to be signed by submission to portal? Or the cross signing certificates sufficient for user mode drivers to get signed and install on windows 10 with secure boot ON? 

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 1eb19d33-8f4c-f379-b645-938a746c5615
* Version Independent ID: 8efb9fb4-1e9a-1a8a-994b-450495449e15
* Content: [Driver Signing Policy](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/kernel-mode-code-signing-policy--windows-vista-and-later-)
* Content Source: [windows-driver-docs-pr/install/kernel-mode-code-signing-policy--windows-vista-and-later-.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/kernel-mode-code-signing-policy--windows-vista-and-later-.md)
* Product: **windows-hardware**
* GitHub Login: @tedhudek
* Microsoft Alias: **windowsdriverdev**

nirvantosh, 2018-07-13:
To be more precise, what about the drivers that does not have any kernel component of their own, for instance a virtual printer driver which is not associated with any hardware and does not have any custom kernel component, do we need to go through portal submission for that also, or a cross signing certificate is sufficient for windows 10 secure boot ON?

tedhudek, 2018-07-13:
Hi @nirvantosh, there is no signing requirement for user mode drivers.  Does that answer your question?

nirvantosh, 2018-07-14:
Hi ted, 

well I tried installing a sample print driver "XPSDrv Sample driver" from WDK samples to windows 10 and it didn't install until I disabled 'Driver Signature Enforcement' which means it does need a sign to get installed. And it is a user mode driver as it just generates a XPS file on print and has no kernel module of itself.

So, as much I understand based on your response and my experience, you are saying I do not need to go through portal and EV signing but I still need to sign my user mode driver using standard code signing certificate. Is my understanding correct?

KeloCube, 2019-11-26:
> Hi @nirvantosh, there is no signing requirement for user mode drivers. Does that answer your question?

Hi, are you sure this is the case?

According to [this forum answer](https://social.msdn.microsoft.com/Forums/en-US/fa6c4811-f5b2-4351-b32a-09bc5c0386db/must-i-sign-of-a-windows-10-umduser-mode-driver?forum=wdk), user-mode drivers have the same code signing requirements as kernel mode drivers. This is backed by my own experience: after having ported a HID driver to UMDF and signing the driver package, SetupAPI gives the following error during installation: "Error 0x800b0109: A certificate chain processed, but terminated in a root certificate which is not trusted by the trust provider."

If the forum answer is right, could the documentation be updated to reflect this?

nirvantosh, 2019-11-27:
Although, I am not sure how relevant this is but I guess its good to share my experience too. As much I know, the kernel-mode drivers need to go through portal and EV signing i.e. via Microsoft verification compulsorily and won't install if signed with just a standard code signing certificate. 

But this wasn't the case for XPSDrv sample print driver which I believe, thus, is a user-mode driver. I found that the XPSDrv Sample Print driver, which is a virtual printer driver and doesn't have any kernel component, was installed in windows 10 after I signed it with a standard code signing certificate. 
 
So, as much I concluded, the user-mode drivers still need signing to get installed in windows 10 but a standard code signing certificate will do there. While for kernel-mode drivers, one needs to get an EV signing certificate and submit to Microsoft's portal for verification and signing.

I hope this helps.

KeloCube, 2019-11-27:
That's interesting, so you're saying you were able to install the print driver without Microsoft's signature after all? Maybe this has to do with the type of driver being installed?

I actually have an update on my case ([a HID minidriver](https://docs.microsoft.com/en-us/windows-hardware/drivers/wdf/creating-umdf-hid-minidrivers)). I used an EV cert to submit the driver to the hardware dev portal for attestation signing, and now the driver is installed without problems. So at least in my case, the driver does seem to require signing by Microsoft.

One more guess I can throw at this is that in my case, the kernel-mode UMDF reflector (WUDFRd.sys) and function driver (mshidumdf.sys) are listed as "driver files" for my device in device manager, even though they are supplied by the OS. In printer drivers, this seems to not be the case. But this is just speculation on my part, I'm not well acquainted with Windows driver architecture.

Either way, I still think the documentation could be improved to have some mention of UMDF, even if just by adding that _certain_ UMDF drivers do not required signing by Microsoft.

nirvantosh, 2019-11-27:
> That's interesting, so you're saying you were able to install the print driver without Microsoft's signature after all? Maybe this has to do with the type of driver being installed?

Yes, I did not go thru portal submission and no Microsoft signature. But I did sign the code with  standard code signing certificate. And yes, I do think this has to do with driver type. I guess UMDF drivers don't need Microsoft's sign. Or may be this has to do something with hardware and kernel components. As I tried this with virtual printer driver which doesn't need any actual printer hardware. So there was also no way found by me to actually do a submission to Microsoft portal as for printer drivers, it demanded the HW printer thing. 

Since I am too not very well-versed with windows driver thing and processes around, I guess someone who is, should take into account our cases and update the documentation properly.

mirh, 2023-09-26:
> "Error 0x800b0109: A certificate chain processed, but terminated in a root certificate which is not trusted by the trust provider."

[That sounds like](https://learn.microsoft.com/en-us/troubleshoot/windows-server/identity/valid-root-ca-certificates-untrusted) a "normal certificate problem", not in any way specific to drivers. 
Maybe you just signed it with such a lowkey standard certificate that it wasn't even in the trusted publishers list? 

Your interpretation that shipping .sys files "at all" could have influenced the results is interesting, but my bet is that if you just [imported](https://stackoverflow.com/questions/75172173/silent-install-of-the-application-fails-to-install-driver-if-vm-application-is-u) your root certificate it would have worked.