MicrosoftDocs/windows-driver-docs #657: Values for parameter 4

LuigiBruno, 2018-07-15:
Provide examples of the most common values for the parameter 4 of this bug check or a link to a list of the NT status codes that could apply to this error.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: c5fdc587-37f6-7dc7-0f7d-2ddc189a1123
* Version Independent ID: 5b82fd28-1d78-0f87-7605-fc5309e6d1cd
* Content: [Bug Check 0x74 BAD_SYSTEM_CONFIG_INFO](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x74--bad-system-config-info)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0x74--bad-system-config-info.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0x74--bad-system-config-info.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2018-08-20:
@LuigiBruno - I added a link to the NT Status Values exception code list to the BAD_SYSTEM_CONFIG_INFO topic. Thanks for your input on the stop code / bug check docs.

closed 2018-08-20 by DOMARS

LuigiBruno, 2018-09-29:
Thank you @DOMARS.