MicrosoftDocs/windows-driver-docs #664: INF files line continuation with comment

tugrul, 2018-07-16:
https://github.com/MicrosoftDocs/windows-driver-docs/blame/staging/windows-driver-docs-pr/install/general-syntax-rules-for-inf-files.md#L84

What about line continuation before comment?

``CopyFiles = "SomeDirectory\"\ ; comment 
,SomeFile
``

Is it equivalent

``CopyFiles = "SomeDirectory\" ; comment ,SomeFile``

or

``CopyFiles = "SomeDirectory\",SomeFile ; comment ``

?

tedhudek, 2018-07-16:
Hi @tugrul thanks for the question. It is equivalent to `CopyFiles = "SomeDirectory\",SomeFile ; comment`. Text after a semicolon is ignored, so this won't work: `CopyFiles = "SomeDirectory\" ; comment ,SomeFile`.

tedhudek, 2018-07-16:
BTW, I updated the page with this extra case.  It will be live in a few minutes.  Thanks again!

closed 2018-07-16 by tedhudek