MicrosoftDocs/windows-driver-docs #670: Is there a way to pre-download symbols before starting debugging

Torkell, 2018-07-17:
Since the offline symbol packages are no more, the first debug session after each set of Windows updates takes ages as I have to wait while Visual Studio slowly downloads the latest symbols (and suffer through lots of pauses as e.g. opening a file dialog causes more symbols to be downloaded). Is there a tool or command I can run to download all the symbols for my Windows install before I start debugging?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 3fabe759-b55e-91e3-ff40-764310295b5a
* Version Independent ID: 71562214-8b57-b69f-1e02-aff3a7e826e2
* Content: [Download  Windows Symbol Packages for Debugging](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/debugger-download-symbols)
* Content Source: [windows-driver-docs-pr/debugger/debugger-download-symbols.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/debugger-download-symbols.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

EliotSeattle, 2018-07-17:
Thanks, @Torkell 
@DOMARS is there any workaround for this?

deakjahn, 2018-07-20:
Start debugging, go into Options > Debugging, press Load all symbols (it will be enabled now) and wait for it once. Then uncheck the Microsoft Symbol Server in the upper list to disable it.

cyren-oschneider, 2018-07-31:
@deakjahn but is there a way to achieve that _without_ having to start debugging?

And in particular is it possible to achieve some functional equivalent of what we previously got by downloading the symbol packages? I.e. get symbols for all the system components on the local system?!

[This article](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/using-a-manifest-file-with-symchk) describes a method that could be used in conjunction with some basic scripting. But it still seems like a tool shipping as part of the WDK or SDK could be equipped with the knowledge of what constitutes "system components" (and therefore used to be included in symbol packages) and provide that.

But my guess here would be that MSFT is trying to avoid this, because of the extra traffic or something like that?!

deakjahn, 2018-07-31:
I don't think you have any other way but this SymChk.

Torkell, 2018-07-31:
@deakjahn: I tried your suggestion - unfortunately that downloads the symbols only for those libraries that are currently loaded. If while debugging I then do something that causes other libraries to be loaded (a good example is causing the Open File dialog to appear) then I have to wait while the symbols are downloaded for the additional libraries (or if I've disabled the symbol server then I don't get any symbols for them).

(note that my normal setup is a bit different: I left the list in Visual Studio empty and instead have configured _NT_SYMBOL_PATH with something like `C:\Symbols\Windows10x64;C:\Symbols\Windows10x86;srv*C:\Symbols\OnlineCache*https://msdl.microsoft.com/download/symbols` so it tries the locally-installed symbols first. I did add Microsoft Symbol Servers to the list in Visual Studio when trying your suggestion though but it didn't help)

@cyren-oschneider: I'll investigate that article, thanks.

DOMARS, 2018-07-31:
I'm checking with the dev team to see if we can provide a recommended work flow.
One tool to be aware of is the Get-WindowsUpdateLog power shell command.
https://docs.microsoft.com/en-us/powershell/module/windowsupdate/get-windowsupdatelog?view=win10-ps

cyren-oschneider, 2018-08-01:
@DOMARS could you please elaborate regarding `Get-WindowsUpdateLog`?

The symbol server argument (`-SymbolServer`) for this cmdlet only allows to make sense of the ETL file.

But I don't think @Torkell was after debugging the Windows update process. Rather, as I understood the question, Windows updates regularly **replace system components which** subsequently leads to the previously downloaded symbols being outdated and therefore **triggers another fetching of symbols from the symbol server(s)**.

DOMARS, 2018-08-06:
@Torkell - I worked with the dev team to layout a few options for managing symbols in an offline system when working with Windows update ETL logs.

[Offline Symbols for Windows Update](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/symbols-windows-update)

One important note covered in the docs is that the on and offline systems, must be running the same version of Windows - use winver or ver to confirm this. 

It is recommend using a single online machine per OS build version so the WU symbol cache will build month-by-month and contain the WU symbols from multiple update releases to completely parse the update log.

closed 2018-08-06 by DOMARS

Torkell, 2018-08-06:
@DOMARS the article you linked implies that it will only download symbols required for debugging Windows Update itself, which is not the use case I've got (sorry if I gave you that impression).

My use case is debugging software that I'm developing in C# and C++ using Visual Studio 2017. Having the relevant Windows symbols available aids debugging as then I get accurate stack traces to work with. Whenever my computer updates to the next major Windows 10 release, the first time I try to debug my program I have to wait for a long time as all the updated symbol files are downloaded. I then encounter further pauses as my program loads more libraries and has to download the symbols for those.

Previously with the offline symbol packages I could largely avoid this by downloading the offline package, starting the install, and then getting on with some other work while the installer completed. I could also send the installer to my colleagues so they don't have to wait for it to download from the internet.

aluhrs13, 2018-08-07:
There are two was you can improve your experience a bit here. @deakjahn's suggestion of SymChk is our recommended approach. You can do something like `symchk /r C:\Windows\System32\ /s srv*D:\sym*http://msdl.microsoft.com/download/symbols` to start to cache symbols locally which should get the important Windows DLLS.

If the situation is you and your colleagues, I'd highly recommend setting up a [symbol proxy](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/symproxy). This will let you all effectively share a cache, so once one of you has accessed a symbol, the rest of you will get it much quicker.

cyren-oschneider, 2018-08-10:
Ooh, that's a clever trick with `symchk /r` ... should have thought of that myself.

In fact at our company I've set up a symbol cache simply as Samba share. Since everyone was pulling via that cache (thanks to `_NT_SYMBOL_PATH`) it was acting almost like a proxy. But I think I can see the advantage of setting up a proper proxy. My biggest issue with _that_ is: IIS.

I had to work with IIS over a decade ago and I did _not_ enjoy it. Back then I would use Apache whereever it was possible and IIS wasn't the only option. In the Windows ecosystem, unfortunately, it often is the only option. These days I use Nginx. Simple, powerful, fast. None of which are attributes I'd apply to IIS 😁 

Anyway, what I meant to say with my above rant was that it would be way cooler by Microsoft to document what their SymProxy does to the extent that I could sit down and write a Python script or an Nginx module and it would work.

I know, I know. The path conventions on the symbol servers are relatively easy, but if there's no specification there's always something that is rarely used that you're bound to miss when mimicking the original.