MicrosoftDocs/windows-driver-docs #671: Links to "Microsoft OS Descriptors" and "Microsoft OS Descriptors Specification" are not working.

wes-b, 2018-07-18:
Links to "Microsoft OS Descriptors" and "Microsoft OS Descriptors Specification" are not working.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 80765aec-291f-b13d-ff28-10e1afa6b742
* Version Independent ID: 016a3903-ba16-1ea6-3804-ab4e6318444e
* Content: [WinUSB Device](https://docs.microsoft.com/en-us/windows-hardware/drivers/usbcon/automatic-installation-of-winusb#winusb-device-installation-by-using-the-in-box-winusb-inf)
* Content Source: [windows-driver-docs-pr/usbcon/automatic-installation-of-winusb.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/usbcon/automatic-installation-of-winusb.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-07-18:
Thanks, @wes-b 
@PageWriter-MSFT the FWLink points to http://www.microsoft.com/whdc/connect/usb/os_desc.mspx for the descriptors. There's content at https://msdn.microsoft.com/en-us/windows/hardware/gg463179.aspx, but I am unjclear if that and the spec link are ported over. Can you please fix this in this topic and in usbcon/microsoft-defined-usb-descriptors?
Thanks!

PageWriter-MSFT, 2018-07-19:
Hello,

Thanks for reporting the broken link. This is now fixed.

Page Writer

closed 2018-07-19 by PageWriter-MSFT