MicrosoftDocs/windows-driver-docs #676: load dbgeng.dll error

heruix, 2018-07-21:
SymSetDiaSession Can not find in dbgeng.dll ，so can‘t open windbg preview！

---
#### 文档详细信息

⚠ *请勿编辑此部分。 docs.microsoft.com ➟ GitHub 问题链接必须具有此部分。*

* ID: c1e093f9-7df0-17ab-71ae-8af9003ee041
* Version Independent ID: 55ed6fd3-a7a6-f7d5-2a8e-fbf6cc6e2c7b
* Content: [Debugging Using WinDbg Preview](https://docs.microsoft.com/zh-cn/windows-hardware/drivers/debugger/debugging-using-windbg-preview)
* Content Source: [windows-driver-docs-pr/debugger/debugging-using-windbg-preview.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/debugging-using-windbg-preview.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

heruix, 2018-07-21:
[image]

heruix, 2018-07-21:
[image]

rzenko, 2018-07-25:
any fix for this? i get the same errors

rzenko, 2018-07-25:
[image]
[image]

DOMARS, 2018-07-27:
I checked with the dev team and they are looking into this.

rzenko, 2018-08-02:
Ahh you are right. I feel like a noob!
Renamed C:\Windows\System32\dbghelp.dll to C:\Windows\System32\_dbghelp.dll and now everything is fine

DOMARS, 2018-08-16:
Closing as it looks like the issue has been identified and a work around exists.

closed 2018-08-16 by DOMARS

karstenBriksoft, 2018-11-16:
this problem still exists today. the workaround to replace the file in system32 is impractical due to window's security restrictions. i.e. I can't move, rename or replace the file in system32. It's also surprisingly complicated to access the WinDBG installation in first place.

ostefano, 2018-11-19:
I agree. This is still broken, and the workaround is not a silver bullet.

WeaponizedMeme, 2018-12-04:
Why the hell is this still not fixed? Jesus fucking christ.

lctrcl, 2018-12-04:
Yeah, very frustrating, fresh Win10 installation and VMware Fusion, same error and the workaround doesn't work as well, after replacing `dbghelp.dll` in `system32` system stopped booting at all.

kbandla, 2018-12-18:
Just ran into this issue on a new windows VM, and I needed a workaround. The problem seems to happen because of VMWare's SVGA 3D drivers, which are part of the VMware Tools that get installed on VMs:
> C:\Windows\SYSTEM32\vm3dum64_loader.dll
> C:\Windows\system32\vm3dum64.dll

These DLLs seem to load the %SYSTEM% dbg dll, causing the issue. 

## Workaround 

So instead of touching the DLLs in the %SYSTEM% folder, 
* copy the contents of the Windbg Preview App, to a new directory
* copy the appropriate `dbghelp.dll` to the root of this new directory from the `amd64` or `x86` directory 

In a new command window:
```
cd Desktop
mkdir windbg_app
robocopy "C:\Program Files\WindowsApps\Microsoft.WinDbg_1.1811.26004.0_neutral__8wekyb3d8bbwe" windbg_app /s /e
cd windbg_app 
```
Then, if you are on an x64 system:
`copy amd64\dbghelp.dll .`
Or if you are on an x86 system:
`copy x86\dbghelp.dll`

That is it. You should now be able to run `DbgX.Shell.exe` from the new directory (windbg_app in the above example)

DOMARS, 2018-12-18:
@aluhrs13 - Possible alternative work around for VMWare dbghelp.dll issue. Not sure updates will work with the moved app directory on the desktop.

shankar-ray, 2019-02-08:
> Confirm the problem still exists.
> Most hassle-free workaround I found so far was disabling VMware's 3D graphics acceleration from the VM's Display hardware settings.

@ibookstein you are a Genius!

ocitrev, 2019-02-19:
I had the same problem and it was caused by NVIDIA GeForce Experience.  I uninstalled and WinDbg Preview works fine now.

xslattery, 2019-02-23:
I can confirm, uninstalling NVIDIA GeForce Experience, fixes it for me.

craniumrat, 2019-02-23:
Can confirm. I just uninstalled NVIDIA GeForce Experience. Windbg Review fired right back up.

lovdov, 2019-02-27:
It seems that recent nvidia update fucked windows up.
Nice job Nvidia as always

lwahonen, 2019-03-02:
WinDBG only works if vmware tools are uninstalled, which makes for a pretty sad user experience.

Ch40zz, 2019-03-10:
Simple fix application without any error checking.
Feel free to add error handling yourself:

```
STARTUPINFOA si = { sizeof(STARTUPINFOA) };
PROCESS_INFORMATION pi = { 0 };
CreateProcessA("C:\\Program Files\\WindowsApps\\Microsoft.WinDbg_1.1902.7001.0_neutral__8wekyb3d8bbwe\\DbgX.Shell.exe", 0, 0, 0, FALSE, CREATE_SUSPENDED, 0, 0, &si, &pi);

char szPath[] = "C:\\Program Files\\WindowsApps\\Microsoft.WinDbg_1.1902.7001.0_neutral__8wekyb3d8bbwe\\amd64\\dbghelp.dll";
void *dllPath = VirtualAllocEx(pi.hProcess, 0, sizeof(szPath), MEM_COMMIT | MEM_RESERVE, PAGE_READWRITE);
WriteProcessMemory(pi.hProcess, dllPath, szPath, sizeof(szPath), 0);
HANDLE hThread = CreateRemoteThread(pi.hProcess, 0, 0, (LPTHREAD_START_ROUTINE)LoadLibraryA, dllPath, 0, 0);

WaitForSingleObject(hThread, INFINITE);

typedef NTSTATUS (__stdcall *tNtResumeProcess)(HANDLE ProcessHandle);
tNtResumeProcess NtResumeProcess = (tNtResumeProcess)GetProcAddress(GetModuleHandleA("ntdll.dll"), "NtResumeProcess");
NtResumeProcess(pi.hProcess);
```

What does this code do?
It starts Windbg Preview in suspended state, then injects dbghelp.dll before any other software such as GeForce Experience can inject their dll to load the wrong dbghelp from System32.
Lets hope this gets fixed some day without the need of using DLL injection lol

WeaponizedMeme, 2019-03-18:
It is now 2019 and this bug is STILL not fixed. People now need to debug Microsoft's debugger? Is that really what this has come to?

goabout2, 2019-03-20:
It is now 2019 and this bug is STILL not fixed. People now need to debug Microsoft's debugger? Is that really what this has come to?

eraser2021999, 2019-03-24:
2019 here and we still have this issue. :( My workaround was to disable Gamestream from Nvidia GeForce Experience.

0x43434343, 2019-04-13:
Apr 13th , still has the same problem, any update for that?

theevilbit, 2019-06-13:
> Closing as it looks like the issue has been identified and a work around exists.
Workaround is not a fix, please reopen.

aluhrs13, 2019-06-13:
> > Closing as it looks like the issue has been identified and a work around exists.
> > Workaround is not a fix, please reopen.

This repo is for tracking documentation issues, not product issues. The correct GitHub issue tracking this is at - https://github.com/microsoftfeedback/WinDbg-Feedback/issues/2#issuecomment-495297591