MicrosoftDocs/windows-driver-docs #690: It's a navigation trap!

skycommand, 2018-07-24:
This page (titled "User-Mode Dump Files") has 10 subpages, each containing only one or two paragraphs. Because of their small sizes, the TOC feature of this site is not used at all. It would be much better if all of it was merged in this page. I'm considering whether a pull request can show what it would look like.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 1abadaf5-ed0e-26fe-1e05-98a32b39e905
* Version Independent ID: 510626db-4304-8d4f-f4db-b81cbe53ccd1
* Content: [User-Mode Dump Files](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/user-mode-dump-files)
* Content Source: [windows-driver-docs-pr/debugger/user-mode-dump-files.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/user-mode-dump-files.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2018-07-24:
I could certainly see at least combining the last two topic sets together and moving the varieties of dump files to the main topic.

So we would end up with three topics in total:
- Overview/Varieties
- Creating
- Analyzing

We also will need to search for the removed topic names in the debugger project and update those links.
I use the VSCode search in files feature to do this. 
This is worth doing. Let me know if you want to take a shot at this, or I can do this towards the end of this week.

Thanks for the interest in the docs!

skycommand, 2018-07-25:
@DOMARS Well, as long as you think it is a good idea... 😊

I already know the removed topic names.

1. [varieties-of-user-mode-dump-files.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/varieties-of-user-mode-dump-files.md)
   1. [full-user-mode-dumps.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/full-user-mode-dumps.md)
   2. [minidumps.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/minidumps.md)
2. [creating-a-user-mode-dump-file.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/creating-a-user-mode-dump-file.md)
   1. [choosing-the-best-tool.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/choosing-the-best-tool.md)
   2. [cdb-and-windbg.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/cdb-and-windbg.md)
   3. [userdump.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/userdump.md)
3. [analyzing-a-user-mode-dump-file.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/analyzing-a-user-mode-dump-file.md)
   1. [analyzing-a-user-mode-dump-file-with-cdb.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/analyzing-a-user-mode-dump-file-with-cdb.md)
   2. [analyzing-a-user-mode-dump-file-with-windbg.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/analyzing-a-user-mode-dump-file-with-windbg.md)

But before I go about searching and replacing in 2,215 files, tell me, isn't there a redirect syntax? Something that we put in, say, `varieties-of-user-mode-dump-files` that say "if anyone came to this page, automatically load `user-mode-dump-files` instead." That way, the merge cross-section is significantly reduced.

skycommand, 2018-07-25:
SERIOUSLY! The "Comment and Close" button shouldn't be so close to "Comment" button!

DOMARS, 2018-08-02:
@skycommand - I ended up with two topics, but greatly consolidated the content, which I think does make it much easier to consume. I ended up adding some in topic orientation links and doing some other work to have this flow together. I searched for all of the removed topics and updated those links in the other topics. I'm in the process of running a link checker on all of the content and will work to clean up any issues that occur as a result of this topic consolidation. Thanks again for this suggestion and collaboration.

closed 2018-08-02 by DOMARS

skycommand, 2018-08-02:
@DOMARS And I ended up resetting my Windows 10 machine twice and eventually nuking and reinstalling it, failing to debug my problem on my own.

The problem is that the contents under "Analyzing a User-Mode Dump File with WinDbg" and "Analyzing a User-Mode Dump File with CDB" are 99% duplicate sentences. And they don't teach how to analyze. They just tell the reader everything he or she needs to start learning how to analyze.