MicrosoftDocs/windows-driver-docs #696: usb2dbg: Windows can't verify the publisher of this driver software (10.0.15063.0)

wangxfdu, 2018-07-25:
Installed WDK for Windows 10, version 1703
OS: 10.0.16299 Build 16299
When trying to install usb2dbg.inf in C:\Program Files (x86)\Windows Kits\10\Debuggers\x64\usb, it prompts "Windows can't verify the publisher of this driver software.".
Driver version is 06/21/2006,10.0.15063.0.
Issue still exists with windbg driver auto install.
No issue with driver version 14393 copied from other WDK installation.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 3f5d6d55-cb35-6b9e-f8ca-e75f70826dc3
* Version Independent ID: 1d913516-b1e0-e872-c337-e30ed5dffce8
* Content: [Previous WDK versions and other downloads](https://docs.microsoft.com/en-us/windows-hardware/drivers/other-wdk-downloads)
* Content Source: [windows-driver-docs-pr/other-wdk-downloads.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/other-wdk-downloads.md)
* Product: **windows-hardware**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-07-27:
Hi @wangxfdu 
We're looking into this. One question. Did you already have any version of the WDK installed on this machine?
Thanks.

EliotSeattle, 2018-08-12:
The team found no history of any issue like you describe. Since I haven't head anything, I am assuming that this is resolved for you. Feel free to reopen this issue if that's incorrect. Thanks!

closed 2018-08-12 by EliotSeattle