MicrosoftDocs/windows-driver-docs #702: Test 2

duncanmacmichael, 2018-07-26:
This is a second test from a different person on the same page.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 74eee0e9-0e41-3611-c43d-b622b3294f4f
* Version Independent ID: 1299179c-11b3-be4d-f77e-791abbda446d
* Content: [User-Mode Driver Framework Frequently Asked Questions](https://docs.microsoft.com/en-us/windows-hardware/drivers/wdf/user-mode-driver-framework-frequently-asked-questions)
* Content Source: [windows-driver-docs-pr/wdf/user-mode-driver-framework-frequently-asked-questions.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/wdf/user-mode-driver-framework-frequently-asked-questions.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

tedhudek, 2018-08-10:
Test

tedhudek, 2018-08-10:
Test

tedhudek, 2018-08-10:
Hi @duncanmacmichael, thanks for opening this issue. @tedhudek, can you take a look?

closed 2018-08-10 by duncanmacmichael