MicrosoftDocs/windows-driver-docs #707: Canno delete the driver

realtebo, 2018-07-28:
&gt;On the Confirm Device Removal dialog box, click the Delete the driver software for this device

What to do when this options is not present at all?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 963a0877-81d6-57b1-2922-459c9077c76c
* Version Independent ID: 49998df1-e7fd-6ce7-eb49-aa9c37f13cd8
* Content: [Using Device Manager to Uninstall Devices and Driver Packages](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/using-device-manager-to-uninstall-devices-and-driver-packages)
* Content Source: [windows-driver-docs-pr/install/using-device-manager-to-uninstall-devices-and-driver-packages.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/using-device-manager-to-uninstall-devices-and-driver-packages.md)
* Product: **windows-hardware**
* GitHub Login: @tedhudek
* Microsoft Alias: **windowsdriverdev**

tedhudek, 2018-07-30:
Hi @realtebo, is this a Windows-provided inbox driver, or your own?

realtebo, 2018-07-31:
I am facing again an issue with a smartphone that cannot be seen from windows. 

It tells me that the device not send the descriptor. 

It worked in the past. I do not know what happened, but regularly windows 10 loose something about Android smartphones. 

I reinstalled Android official Adb driver without any success.

That the situation. In this case, I suppose, we have no driver associated with the device. 

So windows cannot propose to uninstall it.

The question is still valid, I think, because the user cannot understand the message, and have no information or direction on how to fix the situation.

The problem is not the phone, on Windows 7 is regularly seen , both as MTP and charge only.

I am happy you replied me. I am available to give you any info I missed to provide. 

If you choose to close the issue I will understand, but please pass the problem to whoever can improve this slice of the OS because it is a common, and broadly known, problem. 

Without any explanation and any help, how can we fix it?

Thanks for your work and for allowing us to send feedback.

realtebo, 2018-08-17:
I am here again. Another driver, a wireless dongle, that HAS NOT ' Delete the driver software for this device' OPTION.

What to do in these cases?