MicrosoftDocs/windows-driver-docs #712: Missing Winqual.exe

cruzex100, 2018-07-31:
The Partner cannot download the Winqual.exe file. They need this file to upload the cert to Hardware Devcenter. 

Step 2: Sign and upload your Winqual.exe file
From the Hardware Dev Center dashboard, sign in as an administrator with your Microsoft account.
Download the Winqual.exe file from the Hardware Dev Center dashboard, and sign it with the new digital certificate for your company using the SignTool.
On the Administration page, in the Digital certificates tile, click Upload code for digital certification.
On the Digital certificates page, click Browse to locate and select the Winqual.exe file that has been signed with the correct digital certificate for your company.

The partner is redirected to the below link and its a 404 error:
http://go.microsoft.com/fwlink/?LinkId=393250

EliotSeattle, 2018-07-31:
Thanks, @cruzex100 for pointing this out. I'm reaching out to the tool owner to determine how to get the tool now that Sysdev is deprecated. BTW, did you try the steps in the first section of [Winqual submission tool](https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/winqual-submission-tool--winqualexe-)?
Thanks

EliotSeattle, 2018-07-31:
Also, this Issue was filed as an overall issue. Can you please give me the URL of the page that has the steps that failed? Thank you.

cruzex100, 2018-07-31:
Here is the link:
https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/update-a-code-signing-certificate

The Winqual Submission tool is not needed for this...

EliotSeattle, 2018-08-01:
Thanks, @cruzex100 
I fixed the link. It will be live shortly. In the meantime, you can use the [Updated link](https://prodingestionentity1.blob.core.windows.net/hardware/WinqualSubmissionTool.msi?sv=2017-07-29&sr=b&sig=GdjBd2BQOJp4MDVtB5%2BsIJgfyYdV33S5%2FKlwil0IMkk%3D&st=2018-05-01T10%3A09%3A47Z&se=9999-12-31T18%3A29%3A59Z&sp=r).
Thanks,
Eliot

cruzex100, 2018-08-01:
The Updated link is for the Winqual submission tool. The Winqual.exe and Winqual submission tool are not the same.. Please update the download link for Winqual.exe.

EliotSeattle, 2018-08-01:
Ugh. Sorry. Working on it.

cruzex100, 2018-08-01:
No Problem.. Thanks @EliotSeattle

cruzex100, 2018-08-03:
@EliotSeattle , Any updates on the download link?

EliotSeattle, 2018-08-06:
According to the team, "we don’t have separate winqal.exe, it is the Winqual Submission tool link that we provided in HDC."

closed 2018-08-06 by EliotSeattle