MicrosoftDocs/windows-driver-docs #720: Set up a trace session: Video is out of date

DaveRobSwEng, 2018-08-02:
The video uses an older version of MMA so the UI elements arranged differently and take some effort to find

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: b034fd72-470b-76e3-e080-46cb688242db
* Version Independent ID: 3fde5448-65db-8332-2194-c3b301d0d4d5
* Content: [Capture and view USB traces with Microsoft Message Analyzer](https://docs.microsoft.com/en-us/windows-hardware/drivers/usbcon/capture-and-view-ing-usb-traces-with-microsoft-message-analyzer-)
* Content Source: [windows-driver-docs-pr/usbcon/capture-and-view-ing-usb-traces-with-microsoft-message-analyzer-.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/usbcon/capture-and-view-ing-usb-traces-with-microsoft-message-analyzer-.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-08-02:
Thanks, @DaveRobSwEng 
@PageWriter-MSFT can you scope whether we can update the video or create a workaround? Thanks.

EliotSeattle, 2018-12-13:
Duncan, with PageWriter moving to her new assignment, can you take care of this?

duncanmacmichael, 2018-12-13:
Sure @EliotSeattle, I've created an internal work item to track this. I'll prioritize this with my other documentation tasks and see how we can update the videos.

billlattimer, 2019-11-18:
@DaveRobSwEng, thank you for your comment and apologies for the long timeout getting back to you. 

Microsoft Message Analyzer (MMA) will be retired and its download packages removed from microsoft.com sites on November 25 2019.  There is currently no Microsoft replacement for Microsoft Message Analyzer in development at this time.  For similar functionality, please consider using a 3rd party network protocol analyzer tool such as WireShark. 

More information is available at: https://docs.microsoft.com/openspecs/blog/ms-winintbloglp/dd98b93c-0a75-4eb0-b92e-e760c502394f

closed 2019-11-18 by billlattimer