MicrosoftDocs/windows-driver-docs #721: Link to WDK insider preview is broeken

wind-rider, 2018-08-02:
This page links to https://www.microsoft.com/software-download/windowsinsiderpreviewWDK but that returns a 404 page not found.

---
#### Documentdetails

⚠ *Dit gedeelte niet bewerken. Het is vereist om problemen te koppelen tussen docs.microsoft.com en GitHub.*

* ID: 5930505f-6810-b2a2-86b5-3c678aae7f89
* Version Independent ID: 28315cf8-f06f-23ad-2088-9d51d08efe0d
* Content: [Download the Windows Driver Kit (WDK)](https://docs.microsoft.com/nl-nl/windows-hardware/drivers/download-the-wdk)
* Content Source: [windows-driver-docs-pr/download-the-wdk.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/download-the-wdk.md)
* Product: **windows-hardware**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-08-02:
Hi @wind-rider 
Are you talking about the link in this sentence?
Join the Windows Insider Program to get [WDK Insider Preview builds](https://www.microsoft.com/software-download/windowsinsiderpreviewWDK). 
That works for me.

wind-rider, 2018-08-02:
For me it redirects to https://www.microsoft.com/nl-nl/software-download/windowsinsiderpreviewWDK which gives the 404.

If i manually change the url to  https://www.microsoft.com/en-us/software-download/windowsinsiderpreviewWDK, it works indeed.

It seems that the automatic internationalization does not work here

EliotSeattle, 2018-08-02:
That's interesting. Have to look into why it's inserting the Netherlands locale in the URL or why that breaks it. Thanks!

EliotSeattle, 2018-08-03:
Hardcoded the en-us locale into the link. That should take care of the 404 for all locales. Thanks, @wind-rider

closed 2018-08-03 by EliotSeattle