MicrosoftDocs/windows-driver-docs #722: Can Insider Preview WDK's be installed side-by-side with released WDKs?

wind-rider, 2018-08-02:
Could you please document whether it is ok to install an Insider Preview WDK side-by-side with e.g. the 17134 build WDK?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 34d06a09-bca8-3177-00fd-2195f56c4fb9
* Version Independent ID: 5328b553-ad08-4808-0b14-9720484040ce
* Content: [Installing preview versions of the Windows Driver Kit (WDK)](https://docs.microsoft.com/en-us/windows-hardware/drivers/installing-preview-versions-wdk)
* Content Source: [windows-driver-docs-pr/installing-preview-versions-wdk.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/installing-preview-versions-wdk.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **eliotgra**

EliotSeattle, 2018-08-03:
Thanks, @wind-rider 
You should be fine installing side by side. Before I add that statement to the download page, I am checking with the kit team to make sure there aren't any edge cases that would interfere with development. I have several versions installed and see no issues, but I'm not creating production-level drivers.

hansmbakker, 2018-08-13:
(changed my GitHub handle) Thank you!

EliotSeattle, 2018-08-15:
I have heard of no issues from the team. Best of luck!

closed 2018-08-15 by EliotSeattle