MicrosoftDocs/windows-driver-docs #723: Request for clarification regarding mounting/loading of ELAM hive

cyren-oschneider, 2018-08-02:
At the time of writing the article states (emphasis mine):

> The ELAM hive is unloaded after its use by Early Launch Antimalware for performance. If a user mode service wants to update the signature data, **it should mount the hive file from the file location \Windows\System32\config\ELAM**. The storage and retrieval format of these data BLOBs is left up to the ISV, but the signature data must be signed so that the AM driver can verify the integrity of the data.

What's the best practice here? Generate a GUID, convert that to a string and then use that as key into which to mount the hive (via `NtLoadKey` or `RegLoadKey`)?

I am thinking about the chance for clashes between software agents loading the ELAM hive and trying to populate it. So should we stick with the `HKLM\ELAM` (`\Registry\Machine\ELAM`) name or generate a unique name to be used for mounting/loading the hive?

Or are the chances of this happening negligible and I should ignore this scenario altogether?

Thank you and with best regards,

Oliver
---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: da63a995-c445-b722-73d7-2379d00ad1d5
* Version Independent ID: eef47bcf-ed60-ee4b-9341-fef0c8d1eb1d
* Content: [ELAM Driver Requirements](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/elam-driver-requirements)
* Content Source: [windows-driver-docs-pr/install/elam-driver-requirements.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/elam-driver-requirements.md)
* Product: **windows-hardware**
* GitHub Login: @tedhudek
* Microsoft Alias: **windowsdriverdev**

cyren-oschneider, 2018-08-06:
Hi,

just as an update. For now I decided to play it safe and load the key onto a subkey of `HKLM` that has a unique name (UUID I generated). I am not generating a new UUID every run of my code, but not using the name `ELAM` should already avoid the issues I am suspecting may occur.

Best regards,

Oliver

tedhudek, 2018-09-07:
Thanks @cyren-oschneider for the feedback, apologies for the delay on this one.  What you did works fine.  Company or product name would also be unlikely to clash.  I added a sentence to https://docs.microsoft.com/windows-hardware/drivers/install/elam-driver-requirements with your proposed solution as an example.

Cheers!

closed 2018-09-07 by tedhudek