MicrosoftDocs/windows-driver-docs #731: Can WoA support serial device?

wangmir, 2018-08-07:
Currently I'm trying to build and install virtio device driver for WoA on QEMU, and I successfully installed virtio-blk and virtio-scsi drivers released by RHAT (https://github.com/virtio-win/kvm-guest-drivers-windows), but I failed to install virtio-net (NetKVM) driver because of NDIS issue (https://github.com/MicrosoftDocs/windows-driver-docs/issues/719).

So, I tried to set debugging environment using serial device with windbg on remote PC, but it also was failed.
There are many stories, but in short, I tried not only virtio-serial driver but also isa-serial, pci-serial driver with serial device emulation.
And, in case of virtio-serial device driver, the system shows exception error at installing device driver. On the other hand, when set serial emulation (isa-serial) on booting, the system doesn't show any output on the serial port (works well at UEFI bootloader).

So, I'm curious about "Does WoA has general serial device driver in it?". Can you check this for me? Is WoA ready for serial device? or do you have any other debugging option for device driver on WoA?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: f59b1cca-28f4-a8b8-2ce0-63368297d68f
* Version Independent ID: c2249b1e-b35f-137e-40b8-fbb686b3088f
* Content: [Debugging ARM64](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/debugging-arm64)
* Content Source: [windows-driver-docs-pr/debugger/debugging-arm64.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/debugging-arm64.md)
* Product: **windows-hardware**
* GitHub Login: @tedhudek
* Microsoft Alias: **domars**

DOMARS, 2018-08-23:
@wangmir - I have located the dev team and I'm checking to see what I can find out on ARM debugging with a serial COM port.

DOMARS, 2018-09-14:
@wangmir - Thanks for the question, so far we are not supporting the use of serial com ports for debugging on ARM on VM environments such as QEMU. I passed your input on to the product team and they use input like this to guide future product work. But for now, this is not a supported debugger transport scenario.

closed 2018-09-14 by DOMARS