MicrosoftDocs/windows-driver-docs #733: Supportability

tusharchandwani, 2018-08-07:
Great documentation on the usage of the included tools. But a supportability statement of the tools themselves (troubleshooting while using them) would be good, somewhat akin to the statement provided with all Sysinternals tools.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 162fe2fe-9216-f4c5-9f91-3df8776bb99e
* Version Independent ID: 013dbf6b-0790-9c11-24f4-dbf24d64401e
* Content: [Download Debugging Tools for Windows - WinDbg](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/debugger-download-tools)
* Content Source: [windows-driver-docs-pr/debugger/debugger-download-tools.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/debugger-download-tools.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

EliotSeattle, 2018-08-07:
Thanks for the kind words and the suggestion, @tusharchandwani 
@DOMARS I know you're slammed with pre-release content, but can you please triage the request from @tusharchandwani for a supportability statement in a future milestone? Thanks!

aluhrs13, 2018-08-07:
Can you share an example about what you'd expect? I'm not seeing what you're referring to on any SysInternals pages.

tusharchandwani, 2018-08-08:
https://docs.microsoft.com/en-us/sysinternals/license-faq
>
Q: Is there technical support available for the Sysinternals tools?
A: No. All Sysinternals tools are offered 'as is' with no official Microsoft support. 

https://docs.microsoft.com/en-us/sysinternals/license-terms#support-services

DOMARS, 2018-08-16:
@aluhrs13 - Can you determine if there is information that would be appropriate to provide?

DOMARS, 2018-10-10:
@tusharchandwani - For debugger product input like this, the best route is to follow this link to file product feedback.
http://aka.ms/dexex
 This system allows customers to file and up vote product issues. The link automatically sets the product category to Debugging Tools for Windows. I also passed this request on to the debugger product team.
Thanks again for reaching out to the Windows Driver Developer Documentation team.

closed 2018-10-10 by DOMARS