MicrosoftDocs/windows-driver-docs #734: Question about  deferred module and windbg

haoyun-chaom, 2018-08-08:
I open one dump file with windbg to analyze crash reason. The example.dll I want to check was loaded by the crash process, but all the threads in the crash process did not contain the example.dll and  the symbols of the example.dll was marked as "deferred" when I use "lm" command in windbg. In this case, whether could I confirm that the crash was not caused by the example.dll ?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 5f8b7bd8-4c42-9330-b6c1-4bf43583ebd9
* Version Independent ID: 8188de6f-35f4-cc6d-7bbc-7717ac6f80d2
* Content: [lm (List Loaded Modules)](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/lm--list-loaded-modules-)
* Content Source: [windows-driver-docs-pr/debugger/lm--list-loaded-modules-.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/lm--list-loaded-modules-.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

EliotSeattle, 2018-08-08:
Thanks, @haoyun-chaom 
I am assigning this issue to @DOMARS our debug expert. @DOMARS can you identify if this is a content issue where we need to clarify or if this is about the code @haoyun-chaom is creating. Then remove either the doc-bug or support-request label with your response. Thanks so much!

DOMARS, 2018-08-16:
Thanks for reaching out to the Windows Driver Developer Documentation team. Unfortunately, we don’t have the resources to offer individual technical advice to the broad array of customers who use the driver documentation.
You may have success posting your question in the following forum, where other driver developers and Microsoft engineers can be found. The community there is quite responsive.
•	[Windows Hardware WDK and Driver Development](https://social.msdn.microsoft.com/Forums/windowshardware/en-US/home?forum=wdk)

In you specific situation, any specific crash file may not contain what is necessary to attribute fault to a specific DLL. If you can reproduce the crash on demand, you could investigate taking a time travel trace using Windbg Preview if that is appropriate for your situation.
[Time Travel Debugging - Overview](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/time-travel-debugging-overview)

Best of luck with this issue.

closed 2018-08-16 by DOMARS

haoyun-chaom, 2018-08-20:
Thanks @DOMARS, I will try Time Travel Debugging