MicrosoftDocs/windows-driver-docs #737: Need clarity on what needs to be run to satisfy coverage of Windows Vista through Windows 10

nwebers721, 2018-08-08:
Under "Signing a driver for all client versions of Windows", step 1 mentions "Run the HLK tests (Windows 10), or the HCK tests (Windows 8.1 and earlier versions)."  

Based on the wording, it is unclear in the step if just using HLK alone will satisfy being able to support everything from Windows Vista through Windows 10, or if you also need to run HCK in addition to HLK.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 1eb19d33-8f4c-f379-b645-938a746c5615
* Version Independent ID: 8efb9fb4-1e9a-1a8a-994b-450495449e15
* Content: [Driver Signing Policy](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/kernel-mode-code-signing-policy--windows-vista-and-later-)
* Content Source: [windows-driver-docs-pr/install/kernel-mode-code-signing-policy--windows-vista-and-later-.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/kernel-mode-code-signing-policy--windows-vista-and-later-.md)
* Product: **windows-hardware**
* GitHub Login: @tedhudek
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-08-08:
@tedhudek can you look at the passage and see how this can be clarified?

nwebers721, 2018-08-08:
In other words, do you actually need to run HLK _and_ HCK to get support for Windows Vista through Windows 10?

EliotSeattle, 2018-08-08:
Hi @nwebers721 
I understand the request. I'm not sure what the correct answer is. @tedhudek is the author of the topic and either knows or knows who to ask. We'll get you an answer. Thanks!

tedhudek, 2018-08-16:
Hi @nwebers721 I've updated https://docs.microsoft.com/windows-hardware/drivers/install/kernel-mode-code-signing-policy--windows-vista-and-later-#signing-a-driver-for-client-versions-of-windows.  Vista is no longer supported so I can't help you there.

closed 2018-08-16 by tedhudek

dorsav, 2018-08-29:
1. Please fix bugs in documentation + HLK vs HCK confusion in most of the places.
2. Suggest proper ways of automating this procedure, because we had to workaround through a lot of things. Almost 2 months.