MicrosoftDocs/windows-driver-docs #741: Missing command to reset gflags

lextm, 2018-08-09:
gflags /i ImageName -ust should be documented as well to reset the environment.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 8240b7f3-d4f1-4f59-a02e-3a22cb713b13
* Version Independent ID: 9b005c38-3997-72a9-9437-38a9b27640c9
* Content: [Using UMDH to Find a User-Mode Memory Leak](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/using-umdh-to-find-a-user-mode-memory-leak)
* Content Source: [windows-driver-docs-pr/debugger/using-umdh-to-find-a-user-mode-memory-leak.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/using-umdh-to-find-a-user-mode-memory-leak.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

EliotSeattle, 2018-08-09:
Thanks for catching this, @lextm 
@DOMARS can you please triage when you can look into this?

DOMARS, 2018-08-17:
@lextm - Thanks for the suggestion, I added information the command to reset the environment and a link to the gflags command options. 
Feel free to use the pencil on the topic to submit a PR for fixes like this in the future.
Thanks again for your interest in improving the docs.

closed 2018-08-17 by DOMARS