MicrosoftDocs/windows-driver-docs #745: A typo

csujedihy, 2018-08-10:
The process object is ~~now~~ no longer available, as the echoapp.exe process is no longer running.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: f5abd01e-3c7e-11a8-a6c6-62bfba0ef8b6
* Version Independent ID: dceb81b3-b20f-91d1-85d0-fa68c762482b
* Content: [Debug Universal Drivers - Step-by-Step Lab (Echo Kernel Mode)](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/debug-universal-drivers---step-by-step-lab--echo-kernel-mode-)
* Content Source: [windows-driver-docs-pr/debugger/debug-universal-drivers---step-by-step-lab--echo-kernel-mode-.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/debug-universal-drivers---step-by-step-lab--echo-kernel-mode-.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

EliotSeattle, 2018-08-10:
Just that one tiny letter? Got it!!
Thanks for the Issue, @csujedihy 
@DOMARS just FYI.

closed 2018-08-10 by EliotSeattle

DOMARS, 2018-08-10:
@EliotSeattle - Thanks!