MicrosoftDocs/windows-driver-docs #751: Document is incorrect

LarryK348, 2018-08-11:
HidRegisterMinidriver cannot be called from UMDF driver.
The call requires use of PDRIVER_OBJECT which is not defined in UMDF headers, only KDMF.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: d4ed427e-a01a-368b-7422-2dea0aec50a3
* Version Independent ID: 1730981a-bdca-ba26-b8a1-301473678bb2
* Content: [Minidrivers and the HID class driver](https://docs.microsoft.com/en-us/windows-hardware/drivers/hid/minidriver-operations)
* Content Source: [windows-driver-docs-pr/hid/minidriver-operations.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/hid/minidriver-operations.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

tedhudek, 2018-08-11:
Hi @LarryK348, thanks for opening this issue. @PageWriter-MSFT, can you take a look?

tedhudek, 2019-04-25:
Hi @LarryK348 thanks for the feedback, apologies for the delay getting back to you.  I had a quick look at the page you reference.  Can you be more specific which sentence of the page you're referencing?  Also note that if you have a specific change to propose, you can click the Edit button on the page and open a PR

tedhudek, 2019-05-30:
Closing due to inaction.  Feel free to reactivate as needed.  Thanks!

closed 2019-05-30 by tedhudek