MicrosoftDocs/windows-driver-docs #753: Some mistake

ahura24, 2018-08-13:
Hi and tanx for your good post, but i think exist several mistake on this post for example at the first of them written "Call FwpsRedirectHandleCreate0 to obtain a handle that can be used to redirect TCP connections. This handle should be cached and used for all redirections. (This step is omitted for Windows 7 and earlier.)" but this function doesn't exist on windows 7 according by https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/fwpsk/nf-fwpsk-fwpsredirecthandlecreate0

and other mistake ...

EliotSeattle, 2018-08-13:
Hi @ahura24 
Thanks for reaching out to us. You used the general issues mechanism to send this, so we don't know which particular topic you're talking about. Can you tell us the URL and what other issues you see in the content? 
Thanks so much!

duncanmacmichael, 2018-08-13:
Hi @ahura24 , is this the page you are saying has issues? [Using Bind or Connect Redirection](https://docs.microsoft.com/windows-hardware/drivers/network/using-bind-or-connect-redirection).

If so, the first issue you're pointing out is explained in the topic. You're correct in that [FwpsRedirectHandleCreate0](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/fwpsk/nf-fwpsk-fwpsredirecthandlecreate0) doesn't exist in Windows 7, which is why the topic says that the step involving that function is left out, or omitted, in Windows 7 and earlier.

If you have more specific feedback for the [Using Bind or Connect Redirection](https://docs.microsoft.com/windows-hardware/drivers/network/using-bind-or-connect-redirection) page, please feel free to use the "Content Feedback" button on the bottom of that page to let us know if there's anything else to clarify. I'll go ahead and close this issue for now since it's in the "general issues" category. Thanks for pointing this out and thanks for your interest in improving the docs.

closed 2018-08-13 by duncanmacmichael