MicrosoftDocs/windows-driver-docs #756: Broken or obsolete link(s)

ClaytonCurtis1947, 2018-08-13:
As of 2018-08-13 the link to "BCD Boot Options Reference" under "Related topics" on the page "Overview of Boot Options in Windows" takes the user to "Programming reference for Windows Driver Kit (WDK)", where there does not appear to be useful information on boot options.  I went down this trail trying to understand the fields / parameters in a Windows 10 Boot Loader

URL for "Overview of boot options"
https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/boot-options-in-windows

URL under label "Overview of Boot Options in Windows"
https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/index

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: bf43d908-3c09-9ef2-2998-2b8d115a4967
* Version Independent ID: 25c9c362-8c20-bcf7-43ab-258249569bab
* Content: [Overview of Boot Options in Windows](https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/boot-options-in-windows#editing-boot-options)
* Content Source: [windows-driver-docs-pr/devtest/boot-options-in-windows.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/devtest/boot-options-in-windows.md)
* Product: **windows-hardware**
* GitHub Login: @tedhudek
* Microsoft Alias: **windowsdriverdev**

tedhudek, 2018-08-13:
Hi @ClaytonCurtis1947, thanks for opening this issue. @EliotSeattle, can you take a look?

EliotSeattle, 2018-08-13:
Thanks, @ClaytonCurtis1947 
I fixed the link. The MSDN file had a title of **BCDEdit** Options Reference and a filename of devtest.**bcd_boot**_options_reference. Looks like someone got them munged a little. Should be live in a little while.

closed 2018-08-13 by EliotSeattle

ClaytonCurtis1947, 2018-08-14:
Verified fix. Thanks so much!!