MicrosoftDocs/windows-driver-docs #757: "Installing WDTF runtime library" - neither option works

charliearcuri, 2018-08-13:
The "Installing WDTF runtime library" outlines 2 processes for installing WDTF:
1) by configuring a remote computer -or-
2) manually

The instructions for configuring a remote computer contain a menu path that I cannot find: "Configure a remote computer for testing and deployment (Driver &gt; Test &gt; Configure computers...). "  Where should I see the Driver menu?  In Visual Studio?  In some other UI?

The instructions for installing manually link say "For information, see WDTF Runtime Library and "Manually installing WDTF runtime library on a test computer (alternative method).""
But clicking on the link, I do not see "Manually installing..." I suspect the link was broken by changes from msdn…. to docs...

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: fb47a79b-7bbd-ec6f-e389-282648ba5657
* Version Independent ID: c3d8c1f1-5c56-5492-9c7c-d3b3e29a66a1
* Content: [WDTF Quick Start](https://docs.microsoft.com/en-us/windows-hardware/drivers/wdtf/wdtf-quick-start-)
* Content Source: [windows-driver-docs-pr/wdtf/wdtf-quick-start-.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/wdtf/wdtf-quick-start-.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

charliearcuri, 2018-08-13:
Oops - the first option does indeed make sense when you have the Visual Studio UI open after installing the WDK.

EliotSeattle, 2018-08-16:
Thanks @charliearcuri for the feedback. Turns out a couple of topics were lost in the migration we performed a couple of months ago. I am working on replicating the content and posting it.

talfash, 2018-10-22:
The link is still missing. Can't install WDTF manually. Any idea where to go to?

EliotSeattle, 2018-10-30:
Hi @charliearcuri and @LatteAno.
I updated the topics last week, added supporting links and tested. The path is updated, too. Please let me know if you still have a problem. Thanks for your patience.

closed 2018-10-30 by EliotSeattle