MicrosoftDocs/windows-driver-docs #759: In this zoom method, how to define Step = 0.1 for 1x ~4x zoom

ccdriver, 2018-08-15:
I just want to give real zoom define to outside, for app. that make sense, I don't want to use Step = 1, make min, max range (1~40). because if use (1, 40), that seems 1x ~ 40x.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 7a80a168-cd22-2c27-01ae-b92223de254f
* Version Independent ID: 579ac632-398c-52e1-b181-4ff1ce8c4063
* Content: [KSPROPERTY\_CAMERACONTROL\_EXTENDED\_ZOOM](https://docs.microsoft.com/en-us/windows-hardware/drivers/stream/ksproperty-cameracontrol-extended-zoom)
* Content Source: [windows-driver-docs-pr/stream/ksproperty-cameracontrol-extended-zoom.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/stream/ksproperty-cameracontrol-extended-zoom.md)
* Product: **windows-hardware**
* GitHub Login: @barrygolden
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-08-17:
@barrygolden Can you please triage this request and determine if this is a doc issue or if we should steer this fine developer to the forums?

ccdriver, 2018-08-20:
BTW, in SensorSimulation.cpp,  avscamera sample:
 "m_ExtendedZoom.Step() = 1;   // This avoids the rounding issue." this Step should be 1 * 65536. if app want to use step 1 for zoom sildbar.
but when set this Step() = 0.1 * 65536,  app can't get this step.

barrygolden, 2021-01-12:
Thanks for reaching out to the Windows Driver Developer Documentation team.  You may have success posting your question and comments about how extended zoom step values are used in the following forums, where other driver developers and Microsoft engineers can be found. The community there is quite responsive.

[Hardware and Application Developers on Q&A](https://docs.microsoft.com/en-us/answers/products/windows#hardware-and-application-developers)

[NTDEV forum at OSR Online](https://www.osronline.com/showlists.cfm?list=ntdev)

closed 2021-01-12 by barrygolden