MicrosoftDocs/windows-driver-docs #764: Developer Content?

chazste, 2018-08-15:
This page is labeled as "Developer Content" but only has general debugging. When I was debugging this STOP code none of the guidance on this page helped as it was wininit.exe that was dying during initialization. 

Also according to !analyze -v the second parameter is defined (0 for process died and 1 for thread died.

This page should most likely list the small number of processes that Windows marks as critical and some guidance on where to find log info for each of them. Wininit.exe logs a ton of stuff using WPP but I could not find where it was going.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: e08bde9f-652d-9e48-cd2a-4fb58d368642
* Version Independent ID: 23f4f562-cdcb-6c04-3ea3-fb8653cec22b
* Content: [Bug Check 0xEF CRITICAL_PROCESS_DIED](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0xef--critical-process-died)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0xef--critical-process-died.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0xef--critical-process-died.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

wdkbot, 2018-08-15:
Hi @chazste, I'm the Windows Driver Docs assign bot. Thanks for your interest in the docs! I'm assigning @DOMARS to this issue, and they'll triage this as soon as they have a moment.

EliotSeattle, 2018-08-16:
Thanks for the feedback and suggestoins, @chazste 
@DOMARS please take a look at this and determine how this fits into your current workload.

DOMARS, 2018-08-23:
@chazste - Thanks for this valid input, I will investigate if we can do the work to expand the developer content in this topic.

DOMARS, 2018-09-06:
@chazste  - I have started the process of enhancing the content for this bug check and have drafted expanded content. It may take a while to catch up with the devs associated with this part of the Windows code for technical review, but this is a necessary step to make sure the guidance is accurate.  

This blog post from Raymond Chen and an OSR thread have some related content.
https://blogs.msdn.microsoft.com/oldnewthing/20180216-00/?p=98035
https://www.osronline.com/ShowThread.cfm?link=252374

chazste, 2018-09-07:
@DOMARS - Thanks for the links. I had not seen the OSR post before. Very interesting. On the surface that one seems to describe a race where the user could have gotten some other kernel mode bugcheck instead. It also confirms what I suspected about this code. There are a lot of ways to get it. Especially since the list of modules that make up "critical" processes is not entirely under Microsoft's control as they may be third party code running in a service. I read a post on the NT debugging blog referencing stop 0xFE which is similar.

Anyway I have been thinking for a few days that it my particular crash cause since it is trivial to reproduce and has a somewhat annoying debug profile. What I did was I set the computer name to a string of 16 legal characters (max is 15) in the registry HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\ComputerName\ComputerName and the value ComputerName, not thinking that the name might be too long. This was mixed into a small set of other changes to the system (disabling some services to attempt to minimize writes to a volume protected by UWF on an embedded system). So the cause/effect relation was not immediately obvious to me. 

Because it is so easy to reproduce and because it throws a couple of curve balls at you, it seems to me this would be an interesting blog post for NT debugging blog.

Thanks again. And I'll make sure to read the updated docs once they get posted.

DOMARS, 2018-09-19:
@chazste - I published expanded content that is better targeted to a dev. The topic now includes a description of parameter 2, which was a gap.  The dev team indicated that many times a user dump is also created before the system bugchecks. If that is available, it may be a better dmp to look at instead of the kernel dump, so I added that information as well. Thanks again for your input on this topic and your interest in improving our docs.

closed 2018-09-19 by DOMARS