MicrosoftDocs/windows-driver-docs #765: DIFx Recommended, But Deprecated

druske, 2018-08-16:
There is this recommendation on the page: _"We recommend that you only use the DIFx tools to create driver uninstallers. Applications that use the DIFx tools to uninstall drivers are designed to work correctly with future versions of Windows."_

This seems contradicted by the deprecation of DIFx [here](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/difx-guidelines), where we read: _"Starting in Windows 10 Version 1607 (Redstone 1), the Driver Install Frameworks (DIFx) tools are no longer included in the WDK. Instead, we recommend providing your driver as a standalone package that doesn't require an installer, ideally through Windows Update."_

Thanks,
Dave

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 0ac742ba-ed23-9086-7e4b-b46f2e0b4e8d
* Version Independent ID: 2bae5d72-094c-5cf5-5134-7ebf60a904e7
* Content: [Using the DIFx Tools to Uninstall Driver Packages](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/using-the-difx-tools-to-uninstall-driver-packages)
* Content Source: [windows-driver-docs-pr/install/using-the-difx-tools-to-uninstall-driver-packages.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/using-the-difx-tools-to-uninstall-driver-packages.md)
* Product: **windows-hardware**
* GitHub Login: @tedhudek
* Microsoft Alias: **windowsdriverdev**

wdkbot, 2018-08-16:
Hi @druske, I'm the Windows Driver Docs assign bot. Thanks for your interest in the docs! I'm assigning @tedhudek to this issue, and they'll triage this as soon as they have a moment.

tedhudek, 2018-08-20:
Removed the whole node, which was still there in error.  Thanks for finding that @druske

closed 2018-08-20 by tedhudek