MicrosoftDocs/windows-driver-docs #766: code omitted in CfgMgr32.h and SetupApi.h

cjd22, 2018-08-16:
Hi, 

When I use the code above on a UWP it does not recognize HDEVINFO and other declarations because it is omitted in setupapi.h and CfgMgr32.h. I think it is the #if WINAPI_FAMILY_PARTITION(WINAPI_PARTITION_DESKTOP) and #if WINAPI_FAMILY_PARTITION(WINAPI_PARTITION_DESKTOP | WINAPI_PARTITION_SYSTEM) lines respectively that are causing this. 

Please advise

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: a6f42f2d-829c-a6f3-b1a2-fd5e1f05a0e1
* Version Independent ID: 0c739981-7b2d-8387-e7f3-cb3222b79461
* Content: [Porting code from SetupApi to CfgMgr32](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/porting-from-setupapi-to-cfgmgr32)
* Content Source: [windows-driver-docs-pr/install/porting-from-setupapi-to-cfgmgr32.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/porting-from-setupapi-to-cfgmgr32.md)
* Product: **windows-hardware**
* GitHub Login: @tedhudek
* Microsoft Alias: **windowsdriverdev**

wdkbot, 2018-08-16:
Hi @cjd22, I'm the Windows Driver Docs assign bot. Thanks for your interest in the docs! I'm assigning @tedhudek to this issue, and they'll triage this as soon as they have a moment.

EliotSeattle, 2018-08-16:
Thanks, @cjd22 
@tedhudek can you please take a look here and see if this is, indeed, a doc issue?

cjd22, 2018-08-22:
Hi,
Has there been any progress with this issue.

tedhudek, 2018-08-22:
Hi @cjd22 sorry for the delay.

HDEVINFO is from setupapi and setupapi is desktop only.  It is intentionally only present in the desktop partition.  A UWP is supposed to run everywhere and if you are using setupapi, then you won't be able to run everywhere.  Which parts of cfgmgr don't work in UWP?  If writing a universal program, you might need to spell out the A or W versions of functions that have those two versions.  Is that what you are hitting with cfgmgr32.h?

cjd22, 2018-08-23:
Hi @tedhudek 
Yes it is a Universal Program. I am including cfgmgr32.h and the cfrmgr32.lib. For example when try to use the struct CM_NOTIFY_FILTER in my UWP I get the error undeclared identifier because it is omitted in cfgmgr32.h. Please advise

tedhudek, 2018-08-23:
Hi @cjd22 

What version of the SDK are you using?  If you update to https://www.microsoft.com/en-us/software-download/windowsinsiderpreviewSDK I think this may be fixed

cjd22, 2018-08-25:
Hi @tedhudek
When I install the WinSDK sent by yourself in the link. It get to the Installing Windows IP Over USB it then rolls back and will not install. Could you tell me what is happening. Thanks

cjd22, 2018-08-27:
Hi,
Any Advice

smadasam, 2018-08-27:
I tried it successfuly with the 17735 SDK.  It looks like the latest SDK is 17738, so that should also work.

Maybe you need to retarget your solution?

[image]

cjd22, 2018-08-28:
Hi Smadasam, Have you got a link to the 17735 SDK.

cjd22, 2018-08-31:
Hi @tedhudek,
I cannot update my SDK it keeps rolling back. Do you have any advice?

tedhudek, 2018-09-07:
Hi @cjd22 I'm really sorry, I don't have anything else to give you on this one.

I recommend trying [Windows Hardware WDK and Driver Development](https://social.msdn.microsoft.com/Forums/windowsdesktop/en-US/home?forum=wdk)
or [ntdev forum](http://www.osronline.com/showlists.cfm?list=ntdev).

Best of luck with this issue!

closed 2018-09-07 by tedhudek