MicrosoftDocs/windows-driver-docs #768: Error in examining then TDR_RECOVERY_CONTEXT structure.

LuigiBruno, 2018-08-16:
The dt dxgkrnl!_TDR_RECOVERY_CONTEXT (with parameter 1) does not work: the output is

*************************************************************************
***                                                                   ***
***                                                                   ***
***    Either you specified an unqualified symbol, or your debugger   ***
***    doesn't have full symbol information.  Unqualified symbol      ***
***    resolution is turned off by default. Please either specify a   ***
***    fully qualified symbol module!symbolname, or enable resolution ***
***    of unqualified symbols by typing ".symopt- 100". Note that   ***
***    enabling unqualified symbol resolution with network symbol     ***
***    server shares in the symbol path may cause the debugger to     ***
***    appear to hang for long periods of time when an incorrect      ***
***    symbol name is typed or the network symbol server is down.     ***
***                                                                   ***
***    For some commands to work properly, your symbol path           ***
***    must point to .pdb files that have full type information.      ***
***                                                                   ***
***    Certain .pdb files (such as the public OS symbols) do not      ***
***    contain the required information.  Contact the group that      ***
***    provided you with these symbols if you need this command to    ***
***    work.                                                          ***
***                                                                   ***
***    Type referenced: dxgkrnl!_TDR_RECOVERY_CONTEXT                 ***
***                                                                   ***
*************************************************************************
Symbol dxgkrnl!_TDR_RECOVERY_CONTEXT not found.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: b6ad4979-df6e-a8de-880b-f7817ea0f19d
* Version Independent ID: 16d6d813-2b3d-6b08-751c-6a3fdbc48f80
* Content: [Bug Check 0x116 VIDEO_TDR_ERROR](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x116---video-tdr-error)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0x116---video-tdr-error.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0x116---video-tdr-error.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

wdkbot, 2018-08-16:
Hi @LuigiBruno, I'm the Windows Driver Docs assign bot. Thanks for your interest in the docs! I'm assigning @DOMARS to this issue, and they'll triage this as soon as they have a moment.

DOMARS, 2018-08-17:
@LuigiBruno - It seems that you may not have symbols for the driver you are working with, as not all driver vendors or Microsoft provide public symbols for all of their code. To better clarify this scenario, I added *this text* to point this out.

Parameter 1 contains a pointer to the TDR_RECOVERY_CONTEXT. As shown in the !analyze output, *if you have symbols for the associated code*, you can use the dt command to display this data.

Thanks for pointing this out and your interest in helping improve the debugger documentation.

closed 2018-08-17 by DOMARS