MicrosoftDocs/windows-driver-docs #769: xxhatemxx@gmail.com

xxhatemxx, 2018-08-16:
[Enter feedback here]

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: ade10500-7d5b-8b93-aac8-25d31f4392a9
* Version Independent ID: 45c79efc-f024-7f8a-55ce-35e9bfa54b6a
* Content: [WinDbg Preview - Start a user mode session](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/windbg-user-mode-preview#launch-app-package)
* Content Source: [windows-driver-docs-pr/debugger/windbg-user-mode-preview.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/windbg-user-mode-preview.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

wdkbot, 2018-08-16:
Hi @xxhatemxx, I'm the Windows Driver Docs assign bot. Thanks for your interest in the docs! I'm assigning @DOMARS to this issue, and they'll triage this as soon as they have a moment.

EliotSeattle, 2018-08-16:
No issue mentioned.

closed 2018-08-16 by EliotSeattle