MicrosoftDocs/windows-driver-docs #770: Wrong comment

beemavishnu, 2018-08-16:
Currently:
-------------------
PHY statistics for two ports operating on the same port can be combined

To be:
------------
PHY statistics for two ports operating on the same channel can be combined.

Need to replace 'port' to 'channel'.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 30f925f1-9b7f-8dc2-567c-74b5725302a1
* Version Independent ID: 8eb5c01f-ced9-3161-d3ed-f9d59f31d571
* Content: [OID_WDI_GET_STATISTICS](https://docs.microsoft.com/en-us/windows-hardware/drivers/network/oid-wdi-get-statistics)
* Content Source: [windows-driver-docs-pr/network/oid-wdi-get-statistics.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/network/oid-wdi-get-statistics.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

wdkbot, 2018-08-16:
Hi @beemavishnu, I'm the Windows Driver Docs assign bot. Thanks for your interest in the docs! I'm assigning @duncanmacmichael to this issue, and they'll triage this as soon as they have a moment.

duncanmacmichael, 2018-08-16:
Thanks for pointing this out, @beemavishnu. I've made the change and the update should be live later today.

closed 2018-08-16 by duncanmacmichael