MicrosoftDocs/windows-driver-docs #771: Spell mistake

beemavishnu, 2018-08-16:
In the statement "Because the IHV repro mode log is much larget than the normal IHV "

larget to be changed to larger.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 2ef3333b-5dd1-8fd7-7dff-82899964f777
* Version Independent ID: 31c9aed1-b1f1-18a9-d623-8ffa5579b486
* Content: [Logging scenarios](https://docs.microsoft.com/en-us/windows-hardware/drivers/network/logging-scenarios)
* Content Source: [windows-driver-docs-pr/network/logging-scenarios.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/network/logging-scenarios.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

wdkbot, 2018-08-16:
Hi @beemavishnu, I'm the Windows Driver Docs assign bot. Thanks for your interest in the docs! I'm assigning @duncanmacmichael to this issue, and they'll triage this as soon as they have a moment.

duncanmacmichael, 2018-08-16:
Oops, that was a typo. Good catch, @beemavishnu. I've corrected it and the change should be live later today. Also, for little changes like this, feel free to submit a pull request if you'd like and we can merge it in for you.

closed 2018-08-16 by duncanmacmichael