MicrosoftDocs/windows-driver-docs #772: Radar inbuilt memory leak detector

td47, 2018-08-17:
I understand that RADAR is an in-built memory leak detector, and some documentation exists for Windows 7, but is present in Windows 10 as well. However, I cannot find any documentation on this feature or functionality, in the Microsoft DOCS areas. As I see a few RADAR_PRE_LEAK64 and similar 1001 events in the event viewer, I want to find more about these, if they are real or false detections, and how it works. Thanks.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 36af6504-4758-d4fd-def3-92bdecc0649d
* Version Independent ID: eb5e203b-89b6-0a76-7721-4366d978a5d9
* Content: [Determining Whether a Leak Exists](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/determining-whether-a-leak-exists)
* Content Source: [windows-driver-docs-pr/debugger/determining-whether-a-leak-exists.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/determining-whether-a-leak-exists.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

wdkbot, 2018-08-17:
Hi @td47, I'm the Windows Driver Docs assign bot. Thanks for your interest in the docs! I'm assigning @DOMARS to this issue, and they'll triage this as soon as they have a moment.

DOMARS, 2018-08-17:
@td47 - Thanks for that input on adding information on the RADAR memory event entries. We use input like this to help set our priorities for the new docs work, when we have resources to do that work. As any new content would be sometime in the future, here are a few resources for your current work.

It is best to have the source code to look at how it allocates and cleans up memory. With the source code, you can use topics such as these to dig into the apps memory usage.

https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/example-10---detecting-a-heap-memory-leak-in-a-process

https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/using-umdh-to-find-a-user-mode-memory-leak

For how Radar works, this channel 9 video provides additional detail.

https://channel9.msdn.com/Shows/Going+Deep/RADAR-Windows-Automatic-Memory-Leak-Detection

You may have success posting your question in the desktop dev forums, where other developers and Microsoft engineers can be found. The community there is quite responsive.

https://social.msdn.microsoft.com/Forums/windowsdesktop/en-US/home?forum=perfmon&brandIgnore=True&sort=relevancedesc&searchTerm=

Again, thanks for the input, and good luck with your issue.

closed 2018-08-17 by DOMARS