MicrosoftDocs/windows-driver-docs #773: Mention about end point product

ekitagawa, 2018-08-17:
This is Eiichi from Docker support.  It's not virus scan, but it may be worthwhile to mention about end point protection too.  Some vendors simply do not support container sometimes.  I documented it here.  https://success.docker.com/article/endpoint-security-for-windows-containers

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 92187041-3608-8112-a0f1-a4211e189a2a
* Version Independent ID: 09ae3e34-6281-927d-8b44-5283d2c8299e
* Content: [Anti-virus optimization for Windows Containers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ifs/anti-virus-optimization-for-windows-containers)
* Content Source: [windows-driver-docs-pr/ifs/anti-virus-optimization-for-windows-containers.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/ifs/anti-virus-optimization-for-windows-containers.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

wdkbot, 2018-08-17:
Hi @ekitagawa, I'm the Windows Driver Docs assign bot. Thanks for your interest in the docs! I'm assigning @EliotSeattle to this issue, and they'll triage this as soon as they have a moment.

EliotSeattle, 2018-08-17:
Thanks, @ekitagawa 
I'll look into this.

lorihollasch, 2019-06-21:
@ekitagawa , sorry for slow action on this issue! Is it the information on this page that you are suggested be added for AV filter writers? [https://docs.docker.com/engine/security/antivirus/](https://docs.docker.com/engine/security/antivirus/)

lorihollasch, 2019-06-25:
@ekitagawa, I am going to close this issue for now. Feel free to reach out to me to let me know what AV filter writers need to know, thanks.

closed 2019-06-25 by lorihollasch