MicrosoftDocs/windows-driver-docs #774: Hi my name is Nono and i have a big problem

nonodev96, 2018-08-18:
My computer crashes and restarts randomly, I followed a trace until I found the error

Kernel-Power ID 41 (63)

BugcheckCode 307 in hex is 59
Bug Check 0x159: HAL_ILLEGAL_IOMMU_PAGE_FAULT

and I do not know how to fix it
pls help me

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 80315c7c-2e03-137a-173c-0428ff5c6c79
* Version Independent ID: 4ea3c8af-dd5c-44fb-30ab-2fd68975a587
* Content: [Bug Check 0x159 HAL_ILLEGAL_IOMMU_PAGE_FAULT](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x159--hal-illegal-iommu-page-fault)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0x159--hal-illegal-iommu-page-fault.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0x159--hal-illegal-iommu-page-fault.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

wdkbot, 2018-08-18:
Hi @nonodev96, I'm the Windows Driver Docs assign bot. Thanks for your interest in the docs! I'm assigning @DOMARS to this issue, and they'll triage this as soon as they have a moment.

EliotSeattle, 2018-08-20:
Hi @nonodev96 
I'm sorry that you're having problems with your computer. Unfortunately, this documentation set is meant for driver developers and people testing drivers. Off the top of my head, it sounds as if you have either installed a bad driver recently, or one of your drivers has become corrupted. But I'm not a tech support person. While my team takes care of the documentation you found here, we are not trained to help people troubleshoot their computer issues. I would recommend that you start with the guide for troubleshooting blue screen errors at https://support.microsoft.com/en-us/help/14238/windows-10-troubleshoot-blue-screen-errors.

I sincerely hope that you find the solution for this shortly.

closed 2018-08-20 by EliotSeattle