MicrosoftDocs/windows-driver-docs #775: EDID blocks

flatulation, 2018-08-18:
i'm trying to craft a custom driver to fix a monitor with a corrupted edid (wrong checksum, everything else looks fine though when i parse it) and i'm confused over what is meant by EDID blocks. does this mean the individual parts of the 128 byte edid1.3 block, or the whole 128 byte block? does it include the magic 8 bytes 00 ff ff ff ff ff 00 or not?

as an aside, when is this parsed? is it read when the monitor is configured by the video driver? i've heard intel refuses to load the override but nvidia drivers will.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 55e79e97-9c23-a7a4-f7e9-748525022e9c
* Version Independent ID: ee067739-1ae0-4bd8-a206-4d36c31f17aa
* Content: [Overriding Monitor EDIDs with an INF](https://docs.microsoft.com/en-us/windows-hardware/drivers/display/overriding-monitor-edids)
* Content Source: [windows-driver-docs-pr/display/overriding-monitor-edids.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/display/overriding-monitor-edids.md)
* Product: **windows-hardware**
* GitHub Login: @nabazan-microsoft
* Microsoft Alias: **windowsdriverdev**

wdkbot, 2018-08-18:
Hi @flatulation, I'm the Windows Driver Docs assign bot. Thanks for your interest in the docs! I'm assigning @mimisasouvanh to this issue, and they'll triage this as soon as they have a moment.

mimisasouvanh, 2018-09-26:
Hello @flatulation ,

Please look in the "EDID format" section of the article [Overriding Monitor EDIDs with an INF](https://docs.microsoft.com/en-us/windows-hardware/drivers/display/overriding-monitor-edids), this paragraph answers your first question:
"_To update EDID info, the manufacturer’s INF specifies the number of the block to be updated and provides 128 bytes of EDID data to replace the original block. The monitor driver obtains the updated data for the corrected blocks from the registry and uses the EEPROM data for the remaining blocks._"
So basically, EDID blocks is specified by the manufacturer's INF. 
As for the second question, the matter of when the EDID is parsed could change over time. 
I hope these answers help!

Mimi

closed 2018-09-26 by mimisasouvanh